Anonymousdog/displaycameras #2: Screen Flashing/Blanking/Restart?

mcmcomput, 2018-07-15:
**Describe the bug**
The PI boots up fine, starts loading cameras approx 5 sec apart then after about a minute the screen goes black like it lost connection, sometimes flickers once or twice over a period of a minute or two before the camera service restarts i think and re-opens the cameras one by one. I may just have something configured wrong, but it seemed pretty easy to figure out between the two configs. Both are attached for reference.

**Raspberry Pi (please complete the following information):**
 - Hardware model: Raspberry Pi 3 Model B Plus Rev 1.3
 - OS: 2018-06-27-raspbian-stretch-lite / Linux version 4.14.52-v7+
 - gpu memory split: 256MB
 - displaycameras Version: 0.8.3.3

**Display Configuration
 - Display/Monitor resolution: 1920x1080
 - Window matrix: 2x2 (3 cameras)
 - Feed resolution: 640x360

**Camera information (please complete the following information):**
 - Number of cameras: 3
 - Camera models: Hikvision
 - Unmanaged

**Features enabled (please complete the following information):**
 - Rotation: no
 - Display Blanking: yes (tried both)
 - Display Detection: no

**Files
Attached the log and config files,
[layout.conf.default.txt](https://github.com/Anonymousdog/displaycameras/files/2195963/layout.conf.default.txt)
[displaycameras.conf.txt](https://github.com/Anonymousdog/displaycameras/files/2195965/displaycameras.conf.txt)
[logs.txt](https://github.com/Anonymousdog/displaycameras/files/2195966/logs.txt)

Anonymousdog, 2018-07-18:
Your logs are interesting.  I've never tried camera names that are numeric
only, and omxplayer is objecting to a DBUS address specified in the
omxplayer command switches.  For each camera, the DBUS address is the
default address (which should be totally fine) appended with
".camera_name".  I suspect that numeric camera names must violate some
basic namespace restrictions for the DBUS address. I suspect all is well
until the first watchdog run where players won't return a play status via
DBUS.

Try camera names that aren't numeric, eliminate the extra feeds in the
layout file, and retry.  Please, report back: If numeric names are a
problem, I'll need to include that warning in the documentation.

Thanks,
A

mcmcomput, 2018-07-24:
I changed them to "One Two Three" instead. :) Feeds are holding solid now for 30+ minutes.

Anonymousdog, 2018-07-30:
Fixed with documentation of DBUS namespace restrictions in README.md with commit c61e4c1

closed 2018-07-30 by Anonymousdog