Anonymousdog/displaycameras #3: Datetime overlay in a corner

ephestione, 2018-07-18:
**Is your feature request related to a problem? Please describe.**
My xiaoyi's do not always show the correct time, maybe one of them didn't have internet connection so it starts off from 1970-01-01 0:00, or there's the one that no matter how I set the timezone will always show a time that's 2 hours in the past.

**Describe the solution you'd like**
A textual, basic configurable (size/color/position) overlay showing the current system date and time

**Describe alternatives you've considered**
As of now, I have a php script fetching a screenshot of the current display, add an overlay via GD and then returning it to the HTTP request to the server, but it's molasses, naturally

**Additional context**
Attached is an example of what I get with my PHP script (sorry for the lacking freehand-mouse drawing skills in the censored area)
[image]

Anonymousdog, 2018-07-27:
If you can generate the clock face as a video stream we can play like a
camera feed, I believe we can do this with the "layers" branch.

ephestione, 2018-07-27:
When I first got the idea of adding a clock, I did figure that I could generate a rtsp stream of said clock, but I immediately dismissed it as overkill and too much intricate.
So you say that having a video stream is the only feasible way to overlay a text clock on the quad display?

Anonymousdog, 2018-07-30:
Unless you have a utility that displays a live clock with the configuration
you want, yes.

ephestione, 2018-07-30:
There are several scripts that allow to have a live digital clock in the form of a text constantly showing on the screen, for example:
https://www.commandlinefu.com/commands/view/11336/create-a-continuous-digital-clock-in-linux-terminal

maybe there can be a small overlay where a screen instance is used to place a roughly 100px x 20px rectagle in the top-left corner over the top-left camera?

Anonymousdog, 2018-07-30:
Feature enhancement request outside project scope

closed 2018-07-30 by Anonymousdog