Anonymousdog/displaycameras #6: Not working with GPIO based screens

ambanmba, 2018-12-07:
**Describe the bug**
The window that opens with the video stream will not open on a screen that is attached via the GPIO pins. These screens use a driver such as this one to output the video via GPIO instead of HDMI.
https://github.com/goodtft/LCD-show

**To Reproduce**
Steps to reproduce the behavior:
1. Install per README.md
2. Configure my feeds in layout.conf.default
3. Run 'sudo displaycameras start'
4. Image is not displayed on the main screen connected via GPIO and from which the command was run, instead it appears only on the HDMI screen with no way to "move" the window back to the GPIO screen.

**Expected behavior**
Expected the sudo displaycameras start command to display output on the screen connected to the GPIO.

**Screenshots**
N/A

**Raspberry Pi (please complete the following information):**
 - Hardware model B+
 - OS: Raspbian GNU/Linux 9 (stretch)
 - Console: X-Session
 - GPU Memory 64
 - displaycameras Version 0.8.3.1

**Display Configuration
 - Display/Monitor resolution: GPIO Screen: 480x320 / HDMI Screen: 800x480
 - Window matrix: 1x1
 - Feed resolution: 640x360

**Camera information (please complete the following information):**
 - Number of cameras: 1
 - Camera model: Ubiquiti G3 Dome
 - Unmanaged

**Features enabled (please complete the following information):**
 - Rotation: No
 - Display Blanking: No
 - Display Detection: No

**NVR information (please complete the following information):**
 - NVR hardware: None

**Additional context**
N/A

**Files
N/A

Anonymousdog, 2018-12-13:
GPIO screens, being rather small are not really the targeted use case.  Expectation is an HDMI display.

Try to prepend your feeds with "--display=4 "

Anonymousdog, 2018-12-13:
If that throws errors during testing, replace /usr/bin/displaycameras with this version, https://raw.githubusercontent.com/Anonymousdog/displaycameras/layers/displaycameras

ambanmba, 2018-12-14:
Thanks for that... I replaced /usr/bin/displaycameras with the version you suggested (otherwise it was throwing errors).

--display=4 stopped showing the display on the HDMI, but was not showing on the GPIO screen. I tried --display=n where n=1-5 and in all cases except n=4 the image was showing on the HDMI. With n=4 the display wasn't showing anywhere.

Anonymousdog, 2018-12-17:
No idea.  Unless there's some command line parameter we can pass to omxplayer to get it to display video on that screen, you're out of luck.

ambanmba, 2018-12-18:
I'll do a bit of research on this one and get back.

Anonymousdog, 2018-12-18:
Dependency constraint:
omxplayer's --display parameter doesn't support CPIO displays.

closed 2018-12-18 by Anonymousdog