Anonymousdog/displaycameras #8: Delays in Feed

capqwerty, 2019-01-20:
Hi 
I have 4 Unifi cameras, pulling the low res feed. Cameras and Raspbery Pi 3 B+ are all connected via ethernet.
There is no rotation
the config is 
window_positions=(
#First Row
#upper_left
# 640x360
"0 0 639 359" \
#upper_right
"640 0 1279 359" \

#Second Row
#lower_left
"0 360 639 719" \
#lower_right
"640 360 1279 719" \

#off-screen
# 640x360 window just off-screen to the right
"1280 0 1919 359" \
# 640x360 window just below the other
"1280 360 1919 719" \
"1920 0 2559  359" \
)

In the boot config.txt I have set these
disable_overscan=1
gpu_mem=320

The display goes into a delay mode from several seconds up to a reported 10 minutes.

Do you have any suggestions on where to look for the delays?

Anonymousdog, 2019-01-20:
I assume the bold, uncommented lines above aren't actually part of the config file, right?

oceaness, 2019-01-21:
I believe this issue is actually with omxplayer, which displaycameras uses to display the video feeds

popcornmix/omxplayer#647

Anonymousdog, 2019-01-23:
I believe you are correct, oceaness, but I wanted to chase down possible misconfiguration issues with capqwerty before pronouncing that we can't fix it (because the problem lies with the camera/omxplayer combo).

capqwerty, can you post a complete config file?...even with the copy/past weirdness that seems to have pulled out hashes, you haven't posted a complete file.

capqwerty, 2019-01-24:
Config attached
`# This is the camera feed/windows layout configuration file for the
# displaycameras service.  It ONLY configures the layout and feeds for
# the cameras; the rest of the configuration is in displaycameras.conf.
# See the comments in that file for notes on configuring the below.

# This example defines seven 1/2-HD windows, three of which are off-screen to the right,
# through which the service rotates six camera feeds (it actually uses only six windows)
# on a full-HD monitor.  If this suites your needs, modify only the camera names to taste
# and feed URLs to what your cameras or NVR provides.

# Window names

# 2x2 screen with 3 off-screen windows
windows=(upper_left upper_right lower_left lower_right off_screen off_screen2 off_screen3)
# Make sure to account for each window above in the list below.

# Windows positions

window_positions=(
#First Row
#upper_left
# 960x540
"0 0 959 539" \
#upper_right
# 960x540
"960 0 1920 539" \

#Second Row (missing all but the far right window because large_left is double size
#lower_left
# 960x540
"0 540 959 1079" \
#lower_right
# 960x540
"960 540  1919 1079" \

#off-screen
# 960x540 window just off-screen to the right
"1920 0 2879 539" \
# 960x540 window just below the other
"1920 540 2879 1079" \
# 960x540 window just off-screen to the left
"2880 0 3839 539" \
)

# Camera Names

camera_names=(NE SE South SW West Vestibule)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds

camera_feeds=( \
# Mid-Res if your RPi can handle the load
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_1" \
# Low-Res otherwise
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_2" \
#NE
"<Camera stream URL>" \
#SE
"<Camera stream URL>" \
#South
"<Camera stream URL>" \
#SW
"<Camera stream URL>" \
#West
"<Camera stream URL>" \
#Vestibule
"<Camera stream URL>" \
)

# Are we rotating cameras through the window matrix? (default false if not set here)
# rotate="true"`

capqwerty, 2019-01-24:
Hopefully it works as an attachment
[layout.conf.default.txt](https://github.com/Anonymousdog/displaycameras/files/2790488/layout.conf.default.txt)

AOGCrewdawg, 2019-03-19:
I just wanted to jump on this bandwagon. I am seeing the same issue, long (sometimes several minutes) delays.

capqwerty, 2019-03-20:
FWIW I schedule a restart of the service every 4 hours and that brings it back into line

mrosu-sp, 2020-01-14:
Here's how to resolve the posting of code issue [https://github.com/github/markup/issues/541#issuecomment-133881364](url)

logicalmind, 2020-01-14:
I just started getting into this so I don't have any long-term proof yet. But adding the "-r" option to the omxplayer execution seems to keep the latency in check for me. I currently run with the following with good results:

omxplayer -r -n -1 --live --no-keys --no-osd --avdict rtsp_transport:tcp "rtsp://<snip>" --win "x y x1 y1"

You may want to give that a try in your long-term setups and see if it helps.

Anonymousdog, 2020-01-16:
Correct.  That's why this behavior is covered in a prior, closed ticket
with a note linking to the above mentioned (long standing) omxplayer ticket.

This won't be fixed until omxplayer's team fixes it.

Andy

Anonymousdog, 2020-01-16:
Depending on what causes the issue, especially on less expensive cameras (where timings are likely to be non-standard) this may help, yes.

closed 2022-02-16 by Anonymousdog