Anonymousdog/displaycameras #9: Play recorded video files in combination with RTSP streams

mainport, 2019-01-20:
**Is your feature request related to a problem? Please describe.**
I Only have 3 cameras and was wondering if it would be possible to use the 4th screen in the layout to play recorded video files.

**Describe the solution you'd like**
In addition to using omxplayer to open rtsp streams also have the option to use omxplayer to play recorded images or videos

Anonymousdog, 2019-01-23:
It may be possible already...haven't tried.  You could set the relevant camera feed to a video file or playlist and see what happens.  If that doesn't work, it would probably require major changes to how the call to omxplayer is constructed (all the parameters and such).  If so, I doubt I'd try to make that work.

closed 2019-01-23 by Anonymousdog

mainport, 2019-01-23:
Hi,

Thanks for the response, and thanks for creating this! I have solved it with some tweaking.

Is it possible btw to also put in a http stream in combination with a rtsp. I have solved it now by running a second script but i was just wondering.

Again thanks for the great app.

Met vriendelijke groet,

Bas Zelissen

ephestione, 2019-08-15:
> Hi, Thanks for the response, and thanks for creating this! I have solved it with some tweaking. Is it possible btw to also put in a http stream in combination with a rtsp. I have solved it now by [...]

Ahem... would you mind sharing what you did to get those results?