Anonymousdog/displaycameras #10: Run without root-permissions

Luc-Bucher, 2019-01-22:
I recently struggled with running displaycameras as node-child-process alongside another application, which I'd rather not run with root rights, due to security concerns.
Is there a reason, why displaycameras always requires the sudo-prefix?

Is there a feasable solution to that?

As far as I am concerned, executing displaycameras should not need root-rights, as it does not permanently manipulate the system.

Anonymousdog, 2019-01-23:
It's made to run as a service without creating a service account for it.  Thus, root privs are required.  I never cared to look into what, exactly requires root privs.  Not really part of the scope, but I may look into it as I get some time...won't be anytime soon.

Anonymousdog, 2019-02-13:
You can remove the explicit requirement in the script for root by commenting lines 15-18 in the main script and lines 12-15 in omxplayer_dbuscontrol, but the whole self-repairing watchdog feature relies on DBUS registration which is what requires root privs OOTB unless you go out of your way to allow other users DBUS rights.  Since the system is designed to run as a service as root, I won't make this enhancement.

If you can figure out how to allow unprivileged users the necessary DBUS rights, all you have to do elsewise is the above edits.

closed 2019-02-13 by Anonymousdog