Anonymousdog/displaycameras #12: Enable sound on main window

skjeggape, 2019-03-30:
Hello. i just finished setting up displaycams to show streams from a sheep barn. its purpous is to monitor the animals i relations to giving birth. it works wonderfull. having two streams rotating one on-screen and one-off screen. but i miss to have the audio track playing on the stream that is on the main window. hope this is a feature wich can easaly be added?

and pleas forgive my bad non native english.

Hallvar

Anonymousdog, 2019-03-30:
I _think_ I can do this.  Give me a few days.  It'll probably be a set of options: enable_audio, volume_level, audio_stream (in case audio streams aren't the default), audio_output (for jack vs hdmi vs both).  If the first two have values, the system will enable >0 volume on whatever stream is in windows 0 (the first defined, usually top left) and disable it on the rest.

Anonymousdog, 2019-04-02:
Turns out this is a lot more involved than I thought (mostly in the calculating the correct feed on which to turn off sound when rotating it out of the first window), and I want to clean up some changes implemented to allow for 'screen flop' style rotation before attacking this (what ends up being a significant change).  You'll have to give me a few weeks.

skjeggape, 2019-04-04:
Thats fine, im trying to just capture the audio from one stream and have it playing continuisly in the background for now.

skjeggape, 2019-04-08:
Is Therese prehaps some kind of workaround i can use to enable audio on one of my streams? If i can enable allover i can also just remove one mic on a camera to avoid echo. My VLC workaround is sadly a bit unstable.

Anonymousdog, 2019-04-09:
The problem is turning sound on and off when folks are using rotation and want a feed on screen to be the one playing.  I JUST wrote a branch with exactly those edits (although -o hdmi shouldn't theoretically be required) before I saw your edit.

closed 2019-04-09 by Anonymousdog

websystemsoftware, 2019-04-25:
@skjeggape cool this works also for me!!

i have also a client who cannot walk anymore and i must make him a cam outside the house to look his birds....
and is also now fantastic when he can now also listen his birds...

thanks so much
Benny