Anonymousdog/displaycameras #13: Blanking not working on reboot

kris6673, 2019-04-10:
**Describe the bug**
Blanking does not work when you reboot. The black screen does not show, instead it shows the background desktop. If you restart the service with sudo systemctl restart displaycameras, it blanks just fine. status shows that the fbi process is running both when you reboot and restart the service. I have reinstalled the OS and updated everything, didn't work. 
**To Reproduce**
Steps to reproduce the behavior:
Throws no error that i can see since the issue is fixed when you restart the service.

**Expected behavior**
Blanking to blackout the transition between streams.

**Screenshots**
Not really sure if screenshots would help, ask me for the ones you want.

**Raspberry Pi (please complete the following information):**
 - RPI 3 B Vi.2
 - OS: Raspbian Full Stretch
 - Console: Desktop
 - gpu memory split: 256MB
 - displaycameras Version  0.8.3.3

**Display Configuration
 - 1920x1080
 - Window matrix: 1x1 rotating 2 Full HD displays
 - Feed resolution: 1920x1080

**Camera information (please complete the following information):**
 - Number of cameras: 2
 - Camera models : Vivotek FD-8166-F2(W)
 - Managed

**Features enabled (please complete the following information):**
 - Rotation: yes
 - Display Blanking: Yes
 - Display Detection: No

Anonymousdog, 2019-04-10:
I am unable to replicate this behavior on Raspbian_Lite which is definitely the preferred OS for a single use server like this, esp since there is zero need for a resource-hungry GUI on this device.  My bet is the window management bits load long after DBUS and network; so, the service starts before the GUI which plasters itself on a higher frame buffer layer than fbi is running...I don't know.  Bottom line: the fix is running the recommended OS.

kris6673, 2019-04-11:
Sounds likely, switching to lite then. Thanks!

closed 2019-04-11 by kris6673