Anonymousdog/displaycameras #14: fbi replacement

paspo, 2019-04-24:
I set up these scripts in a raspberry running Alpine Linux.

The main advantage is you can cut the power whenever you want, since AlpineLinux is running from RAM. (bonus point: 77MB of used space)

I found an alternative to blanking the screen with fbi: in my startup script I just issue a ```chvt 7``` to change the active virtual terminal to a free one (no console/shell is running by default on that VT).

chmmmr, 2019-04-29:
Hey.  I'd also like to use Alpine Linux.   Is the installation process the same as on Raspbian?

closed 2019-04-29 by Anonymousdog

Anonymousdog, 2019-04-29:
> Hey. I'd also like to use Alpine Linux. Is the installation process the same as on Raspbian?

Alpine Linux appears to user some OpenRC init system; so, I have no idea.  This package is designed for Linux that use systemd.