Anonymousdog/displaycameras #15: Unable to stop screen rotation

lorenzo93, 2019-05-01:
**Describe the bug**
The cameras are continuing to rotate even if rotate is false or the line is commented

**To Reproduce**
Steps to reproduce the behavior:
1. Install per README.md
2. Configure some feeds and keep "rotate" line commented (default behaviour is false)
3. Cameras are still rotating

**Expected behavior**
The cameras should not rotate

**Raspberry Pi (please complete the following information):**
 - Hardware model 3 B+
 - OS: Raspbian Lite downloaded today (2019-05-01)
 - Console: Text Console
 - gpu memory split: 256MB
 - displaycameras commit ac9ad9c

**Display Configuration
 - Display/Monitor resolution: 1920x1080
 - Window matrix: 3x3

**Camera information (please complete the following information):**
 - Number of cameras: 8
 - Camera models: from public stream rtsp://184.72.239.149/vod/mp4:BigBuckBunny_115k.mov

**Features enabled (please complete the following information):**
 - Rotation: no
 - Display Blanking: [yes/no] It happen with both settings
 - Display Detection: yes

If I manually kill after each start rotatedisplays it works like a charm. It looks like the variable in the configuration file is not readed or rotatedisplays starts always

Anonymousdog, 2019-05-01:
I'll try to replicate this bug, but you're much better off actually following the README.md directions which suggest downloading and installing the most recent release, not the current commit (which is always experimental and may have bugs).

jegue, 2019-05-06:
hello, I need some help for creat a layout for layout.conf.large_left for a 1680x1050p resolution monitor.
thanks alot

Anonymousdog, 2019-09-24:
I cannot replicate this issue.

closed 2019-09-24 by Anonymousdog