Anonymousdog/displaycameras #17: turn off/on screen by tapping on it

rventuri76, 2019-06-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Anonymousdog, 2019-06-04:
This package is designed as a non-interactive service.  Neat add-on though.

To add service control, just add lines to stop and start the service 'systemctl start displaycameras' and 'systemctl stop displaycameras' to the relevant events (in addition to the screen backlight control).

closed 2019-06-04 by Anonymousdog

rventuri76, 2019-06-23:
Yes right. Why I din't think it bu myself ;-)
Thanks for the reply.