Anonymousdog/displaycameras #18: Odd issue with flickering

blimeybloke, 2019-06-08:
This is odd.  I have rotation turned on, and using unifi with 7 cameras.  I am doing 4 per screen. 
 Everything loads correctly until its time to roate camera views, the feeds start flickering and overlaying each other.  I am using an older TV that is 1366x768 resolution.  Here is my window position:

`window_positions=(
#First Row
#upper_left
# 640x360
"0 0 682 383" \
#upper_right
"683 0 1365 383" \

#Second Row
#lower_left
"0 384 682 768" \
#lower_right
"683 384 1365 768" \

#off-screen
# 640x360 window just off-screen to the right
"1384 0 1919 359" \
# 640x360 window just below the other
"1384 360 1919 719" \
#"1920 0 2559  359" \
)`

I am sure this issue is me, but just looking for some assistance in case its a bug..
Cheers

jonathanelopez, 2019-06-10:
I had same issue today. Make sure you have your cameras named and listed.. Odd.. But fixed my issue.

Anonymousdog, 2019-09-24:
Camera names are basically the index by which they accessed.  That's why the instructions stress that names are required.

closed 2019-09-24 by Anonymousdog