Anonymousdog/displaycameras #20: Would not stream Sricam SP012

muzzaau008, 2019-06-30:
I have a Sricam SP012 and couldn't get it to stream to the screen even though it worked perfectly on VLC.

Using sudo omxplayer --no-keys --no-osd --avdict rtsp_transport:tcp rtsp://xxx.xxx.xxx.xxx/onvif2 --live -n -1 --timeout 30 would not work either.

I did some research and found out the Scricam SP012 uses udp so I tried sudo omxplayer --no-keys --no-osd --avdict rtsp_transport:udp rtsp://192.168.0.11/onvif1 --live -n -1 --timeout 30 and that worked. But I still wouldn't work in layout.conf.default

So I downloaded the source code and looked through it and found this line(s) in displaycameras:
player="omxplayer --no-keys --no-osd --avdict rtsp_transport:**tcp** --win \"${window_positions[$x]}\" \"${camera_feeds[$i]}\" --live -n -1 --timeout "$omx_timeout" --dbus_name "org.mpris.MediaPlayer2.omxplayer.${camera_names[$i]}" >/dev/null &"
And it includes rtsp transport as tcp, so that's why my stream wouldn't work.

So I added these lines to layout.conf.default :
"# Stream transport types
transport_type=(
#First Row
#upper_left
"udp" \
#upper_right
"tcp" \

#Second Row
#lower_left
"tcp" \
#lower_right
"tcp" \
)

And modified the lines in displaycameras to:
player="omxplayer --no-keys --no-osd --avdict rtsp_transport:\"${transport_type[i$]}\" --win \"${window_positions[$x]}\" \"${camera_feeds[$i]}\" --live -n -1 --timeout "$omx_timeout" --dbus_name "org.mpris.MediaPlayer2.omxplayer.${camera_names[$i]}" >/dev/null &"

So now I can set whether a particular stream is tcp or udp and it will work.

Muzza.

GadgetAngel, 2019-08-13:
I have two raspberry pi cameras having their h264 streams being converted to rtsp via ffmeg and I can not see the rtsp streams with display cameras. Is it possible that ffmeg is translating them to UDP, I think so because motioneyeOS sees them if I use the UDP rtsp format, please allow UDP rtsp streams to be seen by the displaycameras code

Anonymousdog, 2019-09-24:
Request is outside the scope and intent of the project.  It is trivial to alter the code for your use case.  If I were to support this, it would require significant refactoring of the project's code.  That won't happen any time soon (if at all).

closed 2019-09-24 by Anonymousdog