Anonymousdog/displaycameras #21: Using 2nd HDMI on Pi 4

drugfrog, 2019-07-10:
Love the work so much I got the Pi 4, It would be great to utilise the 2nd HDMI port for another display.

Usage Modes I would envisage needed would be:
Mirroring
Current Rotation method spread across two displays
Differing Rotation on each display, eg. 1st rotating Fullscreen, the 2nd static matrix of all feed.

routerino, 2019-07-15:
For that matter, how is the performance on the pi4? I'm specifically thinking about 4k applications, such as having a 4x4 display on a giant 4k output.

Is downscaling improved on the pi 4 as well? We have 4k cameras that I would like to decode and downscale to 1080p (as a single display).

icepicknz, 2019-08-12:
I'd love this too, I have a rpi4 ready to go and would like to do something similar with 1 full screen and 6 matrix on second screen (one large and 5 smaller)

Anonymousdog, 2019-09-24:
I have no idea when I might be able to revisit this for RPi 4's new capabilities.  I suspect that all or most of these requests would require significant reworking and are probably beyond the capabilities of a shell scripted system.  I don't even have an RPi 4.  If I get one, I may revisit this.  Feel free to make code contributions or fork the code.

SebDominguez, 2020-04-19:
Just throwing this here and I don't know if that will be useful but if we had in the feeds configuration file both HDMI 0 (main HDMI output) and HDMI 7 (second HDMI output) we can then send the `--display` option to omxplayer.

This play the feed in the first screen
`omxplayer --no-keys --no-osd --avdict rtsp_transport:tcp rtsp://x.x.x.x:7447/yxzyxzyxz  --live -n -1 --timeout 30 --display 0`

And this play on the second
`omxplayer --no-keys --no-osd --avdict rtsp_transport:tcp rtsp://x.x.x.x:7447/yxzyxzyxz  --live -n -1 --timeout 30 --display 7`

pzzt2t, 2020-06-08:
I'm looking for a solution with Anonymousdog's displaycameras but to output on 2 screens.
It will send 4(2x2) to display1 and 6(3x2) to display2
In total 10 different cameras.
But is it possible?

Currently using displaycamera on different RPI's, and its working very nicely
... just need that last tweak :)

bal0u45, 2020-06-09:
@pzzt2t Using a 2nd display is possible. See the reply above your comment from @SebDominguez.
However, this does require some trial and error especially with camera and display resolutions. I have an RPi4 4GB model running omxplayer to output 6 cameras on a 16:10 aspect ratio monitor and another 2 cameras on a 4:3 monitor. I had to tweak the camera resolution and fps settings quite a bit though. Especially so that there is no downscaling at all, otherwise omxplayer would crash immediately.
You can have a look at [my fork of this repo](https://github.com/bal0u45/displaycameras/tree/multi-display) for a way this could be done with displaycameras. I will finish testing and updating that fork when I have time so that it might be added in a future version.

pzzt2t, 2020-06-09:
@bal0u45 Wow.. sounds very nice.
I look over your repo, not 100% sure what do to.
I run anonymousdog's setup right now.
Do I only need to replace displaycameras with your displaycameras?

or do will I just have to install your modified displaycamera?
.. And im not sure which files needed to modify for the cameras to be displayed on the two different screens.

Sorry for all the questions, hope for you helpfull reply :)

Best Regards

bal0u45, 2020-06-10:
@pzzt2t The fork is still WIP. You would need to include the designated monitor in the layout.conf for each camera you want to display. As I said, I will update the repo and then submit it to @Anonymousdog

sanaking, 2020-12-18:
Hi @bal0u45 !  Thanks for that, I downloaded it, however I am not seeing how to specify which display in the configuration?
Also, I am using two screens with the exact same resolution, so I'm not sure if that would be an issue or not

Anonymousdog, 2020-12-18:
The project doesn't support multiple displays. That's outside the use case.
However,  you may edit the main script code to include an omxplayer target
display parameter (in both places the omxplayer command line is constructed
in the script, lines 121 and 223).  That appears to be '--display 2' for
HDMI 0 and '--display 7' for HDMI 1.

sanaking, 2020-12-24:
@bal0u45 @pzzt2t I have modified displaycameras to display to multiple screens, if you'd like to see my revisions, lmk and I can setup a fork. I basically did what anonymousdog suggested before I read his comment, making modifications to the script and the config file appropriately.

pzzt2t, 2020-12-25:
> @bal0u45 @pzzt2t I have modified displaycameras to display to multiple screens, if you'd like to see my revisions, lmk and I can setup a fork. I basically did what anonymousdog suggested before I read his comment, making modifications to the script and the config file appropriately.

That does sound very nice sanaking.
I would like to try it

Best regards

SvenVD, 2020-12-29:
Hi,

Just want to give a heads up that v3 of Rpisurv has been released https://github.com/SvenVD/rpisurv/releases/tag/v3.0.0-beta3.

This includes full support for dual displays.

Mirroring = possible  by configuring the same streams for each display
Current Rotation method spread across two displays = each display will have his own rotation.
Differing Rotation on each display, eg. 1st rotating Fullscreen, the 2nd static matrix of all feed. = This is possible

Please test it out, we need some beta testers for the new version.

Thanks

Anonymousdog, 2020-12-30:
Nice job!

sanaking, 2021-01-19:
@pzzt2t > That does sound very nice sanaking.
> I would like to try it
> 
> Best regards

I haven't been able to make it stable (it keeps spawning new omxplayer processes which layer on top of each other until it closes them all), so while I could give you the tiny mods that make it work, I haven't figured out how to keep it running or what the issue is.  That said, it may not be a displaycameras issue, because it is also complaining about finding the root and pid file for omx in /tmp and I don't know why, so..
I gave up, and I moved on to rpisurv

closed 2022-02-16 by Anonymousdog