Anonymousdog/displaycameras #22: Step Multiple Displays

pgannett, 2019-07-20:
**Is your feature request related to a problem? Please describe.**
No. The posted software works great! I noticed an extra variable on the online code that would be nice to have in the released code but I was curious if you got it fully working. Specifically the sequence step variable (seq_step)

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I'd like to step 3 displays at a time. I have 5 zones with 3 cameras each so instead of stepping through the available displays 1 at a time I'd like to switch 3 cameras at a time. (ie. display 1-3, then 4-6, 7-9, 9-12, 12-15). I need to step 3 cameras at a time because my cameras are grouped and it'd be useful to see the groups as a whole.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
The displays are stored off-screen. Could I adjust the display to show the pixels off screen? Then keep bumping over into the "extended display" instead of rotating the displays?

**Additional context**
Add any other context or screenshots about the feature request here.
Thanks for any help/advice you can offer!

Anonymousdog, 2019-07-23:
If you pull the current code (vs latest release), do an upgrade install, make that step variable 3, and use a 2x2 grid, it'll do what you want (although it will also display the first feed from the "next" group in the right lower window in addition to the three you want).

closed 2019-07-24 by Anonymousdog

bdschuster, 2019-08-10:
I am not seeing this value? Would i just add `seq_step=2` to the layout.conf.default file or the displaycameras.conf to move 2 cameras at a time? Thanks in advance.

Anonymousdog, 2019-08-14:
Yes