Anonymousdog/displaycameras #23: Raspberry PI 4  video flickering

jeffyl, 2019-08-05:
Not sure anyone has tried a Pi 4 yet. I've got a fresh install of Raspbian Lite, I'm trying to display 2x2 @ 4k (3840x2160) resolution

Using the high quality stream from Unifi-video (1080p), it seems to work fine until the 4th stream comes up, and then I get flickering. This doesn't seem to be the video output flickering, but each video flickers: only 1 video stream shows up at a time. 

Lowering the quality of the videos seems to fix this: 2 medium quality and 2 high quality streams seemed to work fine, upscaled to 1080p to fit the 4k.

Didn't test it for too long though

Ryan-Maughan, 2019-08-15:
Unfortunately I'm having this same issue as @jeffyl. Displaycameras can usually load a few streams before the flickering occurs. I was attempting to output to a 4k TV which at first I thought was the issue, but the issue occurs at 1080p also. Thanks Anonymousdog!

Anonymousdog, 2019-09-24:
What are your all's gpu memory splits?  Has anyone monitored video memory allocation during these tests?

hakanaiseishin, 2019-09-24:
I have tried up to 768 gpu memory split, higher doesn't work. I read about memory split on the official pi forums and it seems to rather not matter on the 3b and later. I have not monitored the GPU memory usage, not even sure how.

ahjcheng, 2019-10-16:
I was experiencing the same issues as well on pi 4. 9 screens on low output to a 1080p TV.

I changed the gpu_mem to 1024 which seem to have solved the problem for me.

https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md

Not sure if this is a fix for everyone but it worked for me.

closed 2019-11-27 by Anonymousdog

shark92651, 2020-01-28:
I am trying to configure a RPi4 with 4GB and a 4K TV that I recently purchased. I am having no luck so far. Just trying to use 2 1920x1080 camera feeds I could only get one to display when I had gpu_mem set to 512. It would switch back and forth (like a flicker, but only displaying one feed at a time) between both feeds until ultimately one would go blank and then the 2nd would display fine. As ahjcheng mentioned above I then tried setting gpu_mem=1024 and I was able to get 2 1080p feeds to display, however an attempt to display 4 of them would only display the original 2. I then tried a lower res substream of 704x480. With this resolution I can only get 7 or 9 to display when trying a 3x3. I'm not sure if this is just poor support for 4k displays in the RPi4, an issue with my 4K TV, or what. 

The documentation linked above states:
**Values of gpu_mem over 512 are not recommended, will provide no performance improvements, and are untested.**

Something did change when I set it to 1024, in that it would allow 2 1080p streams where when I set it to 512 it would flash back and forth between the 2. In the end, it is very inconsistent and I can't figure out any combination of settings that will allow me to display more than 2 streams at 1080p or more than 7 at D1 (704x480).

Anonymousdog, 2020-02-03:
Those gpu_mem split guidelines were written for older hardware (where
settings above 512 reached diminishing returns via stealing too much system
memory).  The RPi 4 has more overall resources; so, higher splits are
fine.  I would be unafraid to try settings as high as half to two thirds of
total memory.  You're one of the first testing this hardware; so, good luck.

Ensure your testing eliminates as many variables as possible by eliminating
as many network hops as you can, running everything on Gigabit Ethernet and
robust switches.  It's pretty easy to saturate a network passing around 4K
video a bunch of times (especially high-bitrate).  You should be able to do
better; I've had three (carefully setup) 1080p streams running briefly on
an RPi 3B+.  Reduce the bit rate on your HD feeds if possible.

What kind of DVR are you using?

Thanks,
Andy

shark92651, 2020-02-04:
I am not  using HD feeds, actually. I am testing with 704x480 substreams and have reduced them to 10 FPS and the lowest bitrate of 256. Despite these changes I cannot display more than 8 feeds.

I am using BlueIris for DVR but I am pulling these RTSP substream feeds directly from the cameras over gigabit network on the same LAN as the cameras.

Anonymousdog, 2020-02-04:
Can you try pulling them from the DVR (for diagnostic purposes).

Andy