Anonymousdog/displaycameras #24: Upper_left not displayed

Jonkheer73, 2019-08-07:
I have 4 cams configured but I only see 3 displayed.
The upper_left won't display.
The feeds are correct because they work in other frames.
I use the Pi 7" screen.
Anyone any idea why this is?
Below my config:

window_positions=(
#First Row
#upper_left
# 800x480
"0 0 399 239" \
#upper_right
# 800x480
"400 0 799 239" \

#Second Row (missing all but the far right window because large_left is double size
#lower_left
# 800x480
"0 240 399 479" \
#lower_right
# 800x480
"400 240  799 479" \

Anonymousdog, 2019-08-14:
Make surebtpnpist code snippets as code, else github autofornats which may add line breaks.  Please repost as such or attach the entire layout config file.

Jonkheer73, 2019-08-14:
I found the problem.
The one that didn't work had a extra symbol in the url. After removal it worked.

closed 2019-09-24 by Anonymousdog