Anonymousdog/displaycameras #31: Performance capabilities

purpleidea, 2019-10-09:
I'm considering a deploy of this with raspberry pi 3b+ and a unifi cloud key gen 2 plus, with their associated cameras. Does anyone know the maximum number of simultaneous streams this can handle?

(1) I don't want to overwhelm the switch or the capabilities of the cameras.
(2) If there was some way to stream _from_ the cloud key v2 instead of directly from the cameras, this might be preferable to avoid each `displaycameras` instance opening N different streams.
(3) Lastly, I'd like to know how many simultaneous different copies of this on different raspberry pi's I can do.

Eg: Is nine cameras and one cloud key v2 with eight separate raspberry pi's each streaming all nine videos possible?

Thanks!

Anonymousdog, 2019-11-27:
15 streams max.
There is only ever one instance of displaycameras running.  Each instance of omxplayer opens one stream.
If the cloud key v2 will generate RTSP streams, then you can use its streams (vs the cameras').
All your other questions depend on your network environment (esp bandwidth) and the nature of your streams (i.e., resolution and bit rate); you'll have to test your own environment (or do the math to figure out a ballpark limitation).
I will say this: IIRC, one RPi running 14 medium-res streams over 100Mbps WAN link took link utilization up to 40% which had a noticeable impact on other operations.  Video is expensive to the network.

closed 2019-11-27 by Anonymousdog

purpleidea, 2019-11-27:
@Anonymousdog Thank you for the reply! As it turns out the Cloud Key V2 takes in all the incoming camera streams, and then the RPI pulls them all from it directly.

It should get exciting, but I'll try and report back with network performance when I can. Cheers!