Anonymousdog/displaycameras #34: Layout Monitor Crashes

Grepster, 2019-11-04:
I have discovered an issue when I attempt to take three feeds and place them next to each other on a single row.  I have tested the issue extensively with the 3x3 layout as well as a customer layout and the key seems to be when the third feed is started the monitor goes signal loss.  It doesn't matter where vertically I place that row.  

When I stack three feeds (row 1, row 2, row 3) it works just fine as long as I don't have three feeds in a single row.  When I stop the service the monitor comes right back.  

I am using a 1920x1080 resolution on a Raspberry Pi 3.

Any troubleshooting tips?  I can't find anything useful in logs.

Anonymousdog, 2019-11-27:
What are you using for a display that you have to manually configure the display resolution?  I run 14 displays 24/7 in my office on a 3x3 layout on a 1080p monitor (with a 3b), and two RPis (1b units, believe it or not) with 6 streams on 1080p monitors (in a large_left layout).

What resolution and bit rate are your feeds?

Post your layout config file if possible.

What does this mean, "This includes a row on the right of the screen."?  Rows are positioned vertically, not horizontally.

Grepster, 2019-11-27:
I meant vertically.  I use a Samsung TV with a display of 1920x1080, I didn't have to set the resolution but I played with various resolutions trying to see if it would resolve my issues.  

The feeds are 1024x576 at 3000 bit rate.

My layout is below:
```
# This is the camera feed/windows layout configuration file for the
# displaycameras service.  It ONLY configures the layout and feeds for
# the cameras; the rest of the configuration is in displaycameras.conf.
# See the comments in that file for notes on configuring the below.

# Window names

# 3x3 screen with 1 off-screen windows
windows=(main upper_right center_right lower_left lower_middle lower_right)

# Windows positions
window_positions=(
#main
"0 0 1279 719" \
#upper_right
"1280 0 1919 359" \
#center_right
"1280 360 1919 719" \
#lower_left
"0 720 639 1079" \
#lower_middle
"640 720 1279 1079" \
#lower_right
"1280 720 1919 1079"
)
# Camera Names
camera_names=(cam1 cam2 cam3 cam4 cam5 cam6)

# Camera Feeds
camera_feeds=( \
rtsp://192.168.1.X:7447/[REMOVED] \
rtsp://192.168.1.X:7447/[REMOVED] \
rtsp://192.168.1.X:7447/[REMOVED] \
rtsp://192.168.1.X:7447/[REMOVED] \
rtsp://192.168.1.X:7447/[REMOVED] \
rtsp://192.168.1.X:7447/[REMOVED]
)

# Are we rotating cameras through the window matrix? (default false if not set here)
#rotate="false"
```
Everything displays fine until I try and include that 6th camera in the bottom right corner.  Playing around with the configs I also found that when I used the 3x3 as soon as the top right corner camera was getting displayed it would cause the issue as well which made me conclude it was related to the three feeds on a single row.

Grepster, 2019-11-27:
Your response made me realize I hadn't tried lowering the resolution (quality) of the feeds.  I replaced the feeds with low quality (640x360) and it worked perfectly.

Grepster, 2019-11-27:
I did some experimenting and determined if I have 3 feeds next to each other on a single row I can only have 1 of the 3 feeds at "medium" quality (1024x576) or I will trigger the issue.

mrosu-sp, 2020-01-14:
I believe I am having similar issues.  I am trying to run a 4x4 grid of  low res 640x360 cameras and when it gets to the third row and second camera, it no longer displays.  After a minute or two, the whole screen goes blank.

A restart of displaycameras starts the camera streams again.  

Here's my layout:

```
#Screen11Kiosk
# This is the camera feed/windows layout configuration file for the
# displaycameras service.  It ONLY configures the layout and feeds for
# the cameras; the rest of the configuration is in displaycameras.conf.
# See the comments in that file for notes on configuring the below.

# This example defines seven 1/2-HD windows, three of which are off-screen to the right,
# through which the service rotates six camera feeds (it actually uses only six windows)
# on a full-HD monitor.  If this suites your needs, modify only the camera names to taste
# and feed URLs to what your cameras or NVR provides.

# Window names

# 4x4 screen, 4 rows and 4 columns
windows=(1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16)
# Make sure to account for each window above in the list below.

# Windows positions

window_positions=(

#First Row
#1
"0 0 479 269" \
#2
"480 0 959 269" \
#3
"960 0 1439 269" \
#4
"1440 0 1920 269" \

#Second Row
#5
"0 270 479 539" \
#6
"480 270 959 539" \
#7
"960 270 1439 539" \
#8
"1440 270 1920 539" \

#Third Row
#9
"0 540 479 809" \
#10
"480 540 959 809" \
#11
"960 540 1439 809" \
#12
"1440 540 1920 809" \

#Fouth Row
#13
"0 810 479 1080" \
#14
"480 810 959 1080" \
#15
"960 810 1439 1080" \
#16
"1440 810 1920 1080" \

)
               
# Camera Names

camera_names=(FLR1HALL FLR15GR FLR15G FLR2GR FLR2HALL FLR2G FLR25GR FLR25G FLR3HALL FLR3GR FLR3G FLR35GR FLR4HALL FLR35G FLR4GR FLR4G)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds

camera_feeds=( \

#FLR1HALL
"rtsp://domain.redacted:7447/redacted" \
#FLR15GR
"rtsp://domain.redacted:7447/redacted" \
#FLR15G
"rtsp://domain.redacted:7447/redacted" \
#FLR2GR
"rtsp://domain.redacted:7447/redacted" \
#FLR2HALL
"rtsp://domain.redacted:7447/redacted" \
#FLR2G
"rtsp://domain.redacted:7447/redacted" \
#FLR25GR
"rtsp://domain.redacted:7447/redacted" \
#FLR25G
"rtsp://domain.redacted:7447/redacted" \
#FLR3HALL
"rtsp://domain.redacted:7447/redacted" \
#FLR3GR
"rtsp://domain.redacted:7447/redacted" \
#FLR3G
"rtsp://domain.redacted:7447/redacted" \
#FLR35GR
"rtsp://domain.redacted:7447/redacted" \
#FLR4HALL
"rtsp://domain.redacted:7447/redacted" \
#FLR35G
"rtsp://domain.redacted:7447/redacted" \
#FLR4GR
"rtsp://domain.redacted:7447/redacted" \
#FLR4G
"rtsp://domain.redacted:7447/redacted" \

)

# Are we rotating cameras through the window matrix? (default false if not set here)
# rotate="true"

			   
```

Anonymousdog, 2020-01-16:
15 cameras is the max.  Your 16 cams will never run. 

That max gets MUCH lower pretty quickly is omxplayer has to downscale feeds to fit windows.  That was to OP's issue, and it's likely yours too (aside from the 16 cams).  Going over around 9 cams also demands setting custom gpu_mem splits and doing hours of bench testing (as memory limits can take minutes to hours to manifest).

closed 2020-01-16 by Anonymousdog