Anonymousdog/displaycameras #35: Camera randomly stops displaying

timleland, 2019-12-06:
I'm using 2 Wyze cameras running RTSP. Everything works great but randomly the cameras go black then come back a few seconds later. Sometimes its just one at a time. Are there any logs to see what could be causing the issue?

Anonymousdog, 2019-12-06:
No. The system doesn't really generate any logs.  You have two
basic options.

   - Replicate what the system is doing by running each feed with omxplayer
   (using a command line that generates logs) long enough to see the behavior,
   or...
   - Modify the main script to include logging options in the omxplayer
   lines.

My bet is a small disruption of connectivity, then the watchdog component
notices the failed feed and fixes it.  This is usually a network problem or
camera issue (with generating a continuous feed) unless you're also doing
something else with the RPi.

Andy

timleland, 2019-12-07:
Thanks for the feedback! These are great tips. I recently upgraded my wifi and it seems to be better. The raspberry pi runs on wireless and I believe it was randomly disconnecting.

closed 2019-12-07 by timleland