Anonymousdog/displaycameras #36: New features and updates

fctr, 2019-12-30:
I made 2 changes in the script that do the following:

1. Replaced fbset with tvservice. tvservice uses actual HDMI output settings

```
...
# If enabled, autodetect display mode, check for layout config file for that
# mode, and load that file if present.
if [ "$displaydetect" = "true" ]; then
        if [ "`grep disable_overscan /boot/config.txt | egrep -v ^#`" = "disable_overscan=1" ]; then
                # This is what I'm talking about
                mode="`tvservice -s | sed 's/.*\], //' | sed 's/ @ .*//'`"
                if [ -r /etc/displaycameras/layout.conf.$mode ]; then
                        . /etc/displaycameras/layout.conf.$mode
                        displaydetectactive=true
...
```
2. Added option force4x4 to calculate and automatically use four windows equally divided on the current output screen.

```
...
else
        . /etc/displaycameras/layout.conf.default
fi

if [ "$force4x4" = "true" ]; then
        mode="`tvservice -s | sed 's/.*\], //' | sed 's/ @ .*//'`"
        vwidth="`echo $mode | sed 's/x.*//'`"
        vheight="`echo $mode | sed 's/.*x//'`"
        vwidthhalf="`expr $vwidth / 2`"
        vheighthalf="`expr $vheight / 2`"
window_positions=(
"0 0 $vwidthhalf $vheighthalf" \
"$vwidthhalf 0 $vwidth $vheighthalf" \
"0 $vheighthalf $vwidthhalf $vheight" \
"$vwidthhalf $vheighthalf $vwidth $vheight" \
)
fi
...
```

ptrtw, 2020-01-08:
may I ask which file to look for and I can add this script?

fctr, 2020-01-08:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/Anonymousdog/displaycameras/issues/36?email_source=notifications\u0026email_token=AOEZ2E32BPISFEUJI5YUUD3Q4WGRJA5CNFSM4KBHBN22YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEILTNAA#issuecomment-571946624",
> "url": "https://github.com/Anonymousdog/displaycameras/issues/36?email_source=notifications\u0026email_token=AOEZ2E32BPISFEUJI5YUUD3Q4WGRJA5CNFSM4KBHBN22YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEILTNAA#issuecomment-571946624",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

Yes.  Hold off for a few, because I'm adding one or two more features. I will email you the entire file when I'm ready.---=> Andrew BurnsSent from the deepest, darkest reaches of my universe. Pretty neat, huh?

fctr, 2020-01-09:
OK, so here's the final update.

I added a few options to make this a little nicer.

[displaycameras.zip](https://github.com/Anonymousdog/displaycameras/files/4037982/displaycameras.zip)

closed 2020-01-09 by fctr

ptrtw, 2020-01-09:
thank u sir :)

fctr, 2020-01-09:
After letting it run for 24 hours, I noticed I missed one small thing. I'll try to post the fixes today or tomorrow.---=> Andrew BurnsSent from the deepest, darkest reaches of my universe. Pretty neat, huh?

[
{

fctr, 2020-01-09:
OK, here's the updated "displaycameras" script. Sorry about that...

[displaycameras.zip](https://github.com/Anonymousdog/displaycameras/files/4042404/displaycameras.zip)