Anonymousdog/displaycameras #38: Can displaycameras handle a 4x4 matrix on 4k monitor with a pi4

shark92651, 2020-01-18:
I'm curious if anyone has been able to go as high as a 4x4 matrix of 16 camera feeds with a RPi4. I have a RPi4 4GB model on order and was hoping I could use it to eliminate a Windows NUC box that runs my current split-screen camera monitor.

RChadwick7, 2020-01-23:
I would be very curious to know the same thing.

Anonymousdog, 2020-01-23:
You can try it.  I know the 15 display limit seemed to have been a limit in
omxplayer and not a resource exhaustion issue; so, I doubt it would work
better on an RPi 4, but don't let me discourage you.

Andy

shark92651, 2020-01-28:
Well I finally had a chance to work on this but I can't even get a 3x3 matrix to work. 8 cameras in a 4x2 is fine, but as soon as I try to display a 9th camera feed in a 3x3 they all start flickering and turning off and on and doing really weird things. I set my memory split to 512, which from my googling is the max value that has been tested, but it did nothing to address the issue with 9 cameras and flickering. Any ideas what to look for? If I could get a 4x3 of 12 cameras working I would be happy.

I am using 704x480 15fps substreams

shark92651, 2020-01-29:
I did some more testing this morning. When changing the gpu_mem to 512 or less, the system will display all 9 feeds in a 3x3 matrix, but has the super-seizure flashing issue. If I up the gpu_mem to 944 or 1024, there is no flashing but will only display 7 feeds.  Any suggestions on settings I should try to resolve the issue?

These are the display related settings that are uncommented in config.txt:
  hdmi_group=1
  hdmi_mode=95
  config_hdmi_boost=6
  dtoverlay=vc4-fkms-v3d
  max_framebuffers=2
  gpu_mem=944

TLovskog, 2020-02-17:
I have the exact same problem with 2560x1440.
With 512 gpu_mem all streams (10) show up, but with some serious flickering. With higher gpu_mem omxplayer fails to start playback of stream 8 and above.
I accidentally run omxplayer with one of the failing streams, to test. Then I started the displaycameras. Now only 6 streams appear and after that opmxplayer fails on all remaining.

... cut successful stream 1-6
Starting omxplayer for Garage
Garage started
Starting omxplayer for LivingRoom
Waiting for LivingRoom omxplayer startup 0
Waiting for LivingRoom omxplayer startup 1
Waiting for LivingRoom omxplayer startup 2
Waiting for LivingRoom omxplayer startup 3
Waiting for LivingRoom omxplayer startup 4
Waiting for LivingRoom omxplayer startup 5
Waiting for LivingRoom omxplayer startup 6
Waiting for LivingRoom omxplayer startup 7
Waiting for LivingRoom omxplayer startup 8
Waiting for LivingRoom omxplayer startup 9
Waiting for LivingRoom omxplayer startup 10
LivingRoom failed playback
... cut

shark92651, 2020-02-17:
I believe the issue is either with Omxplayer or with the RPi4 support for 4k monitors. I tried a different solution that uses Omxplayer and it behaved exactly the same - can only get 8 streams to work.

TLovskog, 2020-02-17:
If I edit the displaycameras script to omit any start of omxplayer between 1-6, number 7 will start. Seams to be something with starting more than 6 omxplayer instances.

Anonymousdog, 2020-02-17:
You can run up to 15 instances, just not at those resolutions.

TLovskog, 2020-02-17:
Thanks for responding.
Resolutions as in my total 2560x1440 or resolutions on my streams 640x460 x 10

rws832, 2020-02-18:
Do you have the code for the 4x4 layout for a 1920x1080 screen? I am trying to get 13 cameras on one screen.

shark92651, 2020-02-18:
I have a code generator that can create it. What size do you want each window to be?

Anonymousdog, 2020-02-18:
The aggregate of the streams, especially if there is ANY downscaling.

Andy

rws832, 2020-02-18:
Thanks for the code. Where can I find this code generator at also? We have several displays to convert over to the displaycameras an it would come in handy.

shark92651, 2020-02-20:
It's a simple C++ app I wrote in Visual Studio. Here is the main code. You just need to add a simple UI or adapt it into a console app. You need to input 4 integers:

seCols.Value and seRows.Value is the size of the matrix (4x4 for example)
seW.Value and seH.Value is the window width and height of the feeds
meResults.Lines is the UI control that receives the output, or you could dump it to a file.

[Form1.txt](https://github.com/Anonymousdog/displaycameras/files/4228167/Form1.txt)

rws832, 2020-02-20:
Thanks Shark92651. This will save a lot of time.

SvenVD, 2020-02-20:
just FYI, https://github.com/SvenVD/rpisurv has integrated autocalculation of the windows. Downside is that currently all streams are handled the same, so you can not make one stream bigger as another, but this covers must use cases like a 4x4.

shark92651, 2020-02-20:
That's good to know. I would give rpisurv a try but I fear it also won't handle 9 streams on RPi4 and 4K monitor since it also uses omxplayer. I have tried two different packages, both based on omxplayer, and both fail to handle 9 streams at 704x480 on 4K monitor.

closed 2022-02-16 by Anonymousdog