Anonymousdog/displaycameras #40: Manual Feed Switching

tedtms, 2020-02-16:
Hello,

Excellent project, thank you so much for this! I'm running an rpi4 attached to a TV with the intent of switching it back and forth between a live camera feed and an encoded RTSP stream from a presentation computer running PowerPoint. So far, the feeds work great!

However, in my use case, I have a specific need to switch from one feed to another manually. The feed should take up the whole display, and rotation isn't really an option for me as I only need it to switch once, and the time isn't predictable. 

Is there a way (or could there be a way) to make displaycameras choose another feed when prompted? So far, I'm considering the quick & dirty solution of having two layout.conf files (1 and 2 for example) and using a script to ssh into the pi, rename the files, restart the service to load the renamed file, then exit. Obviously that's not an ideal solution though.

Thanks!
Ted

Anonymousdog, 2020-02-17:
Not sure this meets your needs, but you can do this out of the box.

Configure for both feeds (with one off screen) and no rotation.

When arbitrary time arrives for the switch, '/usr/bin/displaycameras
rotate' run as root will do the job.  Same command switches it back.

Thanks,
Andy

tedtms, 2020-02-20:
Andy,

Awesome, thank you! That's exactly what I needed.

Cheers!
Ted

closed 2020-02-20 by tedtms