Anonymousdog/displaycameras #45: Camera UDP rtsp stream

pezzotto, 2020-03-31:
Hello and thanks for this great code.

I'm trying to create a display with two cheap Sricam but i noticed that the rtsp stream is UDP instead of TCP.

In fatc if i try sudo omxplayer --no-keys --no-osd --avdict rtsp_transport:**UDP** <camera feed URL> --live -n -1 --timeout 30 
I have no problem to see the image.

There's a way to change the rtsp_transport in the code?

Thanks!

theonemcdonald, 2020-03-31:
Every instance of omxplayer is spawned using the same base command, which uses TCP transport. You can change the transport mode in the code yourself...but again, this changes it for ALL streams spawned by the script

`	player="omxplayer --no-keys --no-osd --avdict **rtsp_transport:tcp** --win \"${window_positions[$x]}\" \"${camera_feeds[$i]}\" --live -n -1 --timeout "$omx_timeout" --dbus_name "org.mpris.MediaPlayer2.omxplayer.${camera_names[$i]}" >/dev/null &"`

pezzotto, 2020-04-01:
Thank you i found it and changed, now it's ok.

The reason is that i have two Sricam that stream an rtsp only udp (strange i know)..

closed 2022-02-16 by Anonymousdog