Anonymousdog/displaycameras #46: GPIO push switch for rotation

8v-piper, 2020-04-13:
Good Day

Thanks for a great program and making it available to us.

I would like to ask if it is possible to use the rotation feature to bring one of the cameras I have displayed on my 4x4 grid, full screen by using a push button connected to the GPIO? 

Seems simple but the changes i made do not seem to generate the desired result.
Any help would be appreciated

Apologies if this is not placed in correct section
Many Thanks

tedtms, 2020-04-14:
Hey @8v-piper, this may help your use case. I asked @Anonymousdog on a previous ticket if manual rotation was possible, and his solution was to run the command `/usr/bin/displaycameras rotate` which would automatically switch the feeds. If you configure one of the feeds off-screen, this should work.

Then, if you configure your GPIO button to just issue that command, it should rotate the feed for you.

Hope that helps!
Ted

8v-piper, 2020-04-14:
HI @tedtms 

Thanks will give it a try i have read up on that ticket and will configure it to send command when it goes high and another one when it goes low again to restore the previous view.

Thanks will update tomorrow after I have tested the solution.

8v-piper, 2020-04-15:
Good Day @tedtms 

I have tried your solution and it does work to rotate.

Unfortunately there is a but, when I rotate the screen it brings the camera it should rotate to in full screen and displays it in the 2x2 matrix.
Then every time you do the command again it just moves it through to another position in the 2x2 matrix until its off screen again.

here is my layout config file settings

# Window names

# 2x2 screen with 1 off-screen windows
windows=(upper_left upper_right lower_left lower_right off_screen)
# Make sure to account for each window above in the list below.

# Windows positions

window_positions=(
#First Row
#upper_left
# 512x384
"0 0 511 383" \
#upper_right
# 512x384
"512 0 1023 383" \

#Second Row (missing all but the far right window because large_left is double size
#lower_left
# 512x384
"0 384 511 767" \
#lower_right
# 512x384
"512 384 1023 767" \

#off-screen
# Full Screen
"1024 0 2047 767" \
)

# Camera Names

camera_names=(NE SE South SW West)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds

camera_feeds=( \
# Mid-Res if your RPi can handle the load
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_1" \
# Low-Res otherwise
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_2" \
#NE
"rtsp://10.0.0.3/axis-media/media.amp?videocodec=h264&resolution=800x600&fps=25&videokeyframeinterval=25&compression=30" \
#SE
"rtsp://10.0.0.5/axis-media/media.amp?videocodec=h264&resolution=800x600&fps=25&videokeyframeinterval=25&compression=30" \
#South
"rtsp://10.0.0.2/axis-media/media.amp?videocodec=h264&resolution=800x600&fps=25&videokeyframeinterval=25&compression=30" \
#SW
"rtsp://10.0.0.7/axis-media/media.amp?videocodec=h264&resolution=800x600&fps=25&videokeyframeinterval=25&compression=30" \
#West
"rtsp://10.0.0.5/axis-media/media.amp?videocodec=h264&resolution=800x600&fps=25&videokeyframeinterval=25&compression=30" \
#Vestibule
)

# Are we rotating cameras through the window matrix? (default false if not set here)
rotate="false"

Am I doing something obviously wrong?
any help is appreciated

Anonymousdog, 2020-04-21:
That method won't change your layout (bringing one feed to full screen).
Within the current paradigm, the only chance of doing what you want would
be triggering a service restart with switch to different layout file (and
that has problems too).

 This use case is well beyond the scope of this project which is intended
for passive viewing.  RpiSurv may come closer to meeting your needs.

Sorry,
Andy

8v-piper, 2020-04-25:
Thank you appreciated I am checking out RpiSurv now.

closed 2020-04-25 by 8v-piper