Anonymousdog/displaycameras #47: Not working

elektronep, 2020-04-15:
Hello. I just want to display one camera. My config looks like this:

windows=(LT2)
window_positions=(
"0 0 1024 768" \
)
camera_names=(LT2)
camera_feeds=( \
"<rtsp://USERNAME:PASSWORD@10.0.0.10:554/mpeg/media.amp>" \
)

My RTSP link works just fine in VLC. Nowhere in the documentation of this program is authentication mentioned so I just following standard form for the link.

I hope someone can explain to me what the issue is. Thanks!

8v-piper, 2020-04-15:
Good Day

I am a new user but I will throw in my 2 cents and hope it helps.

Lets first see if your stream can be played by omx directly before trobleshooting the displaycameras

this is how i checked my own incoming rtsp stream
1. kill the displaycameras service with this command: sudo systemctl stop displaycameras.service
2. Enter your rtsp stream into omx directly:  sudo omxplayer  --avdict rtsp_transport:tcp "rtsp://yourdetailshere"
if it works then we can continue with troubleshooting

elektronep, 2020-04-16:
Yes, this displays the video on the screen!

elektronep, 2020-04-16:
Despite it working on the screen, I get this on my terminal:

Invalid framerate 90000, using forced 25fps and just trust timestamps
Video codec omx-h264 width 1280 height 720 profile 77 fps 25.000000
Subtitle count: 0, state: off, index: 1, delay: 0
Error: Unable to open font
V:PortSettingsChanged: 1280x720@24464.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0

elektronep, 2020-04-16:
I know that the camera has a framerate of 10 fps.

elektronep, 2020-04-22:
Could someone help me?

Anonymousdog, 2020-04-23:
Post your layout.conf.default file as an attachment please.

elektronep, 2020-04-25:
Here it is. I have changed the username and password for privacy and I had to change the extension to be able to attach it.

[layout.txt](https://github.com/Anonymousdog/displaycameras/files/4533759/layout.txt)

Anonymousdog, 2020-04-27:
Correct the syntax in that file (using
https://github.com/Anonymousdog/displaycameras/blob/master/example_layouts/layout.conf.1920x1080.1x1
as
a reference).

Windows names cannot include a space; current config looks like two windows
to the system.
Similar issue with camera names.
Camera feeds shouldn't have a leading "\" (on the first line).

Anonymousdog, 2020-04-28:
> Your suggestion regarding "" conflicts with the readme. I am so
> frustrated. I have followed the readme precisely and it does not work.
> Continues to work just fine when triggered by the omxplayer command. I am
> really about to give up on all of this. Here is my extremely simple file. I
> just want to show one camera all the time.
>
> `windows=(window)
>
> window_positions=(
> "0 0 1023 767"
> )
>
> camera_names=(camera)
>
> camera_feeds=(
> "rtsp://USERNAME:PASSWORD@10.0.0.10:554/mpeg/media.amp"
> )
> `

That looks correct. The file you sent, not so much.

elektronep, 2020-04-28:
I was finally able to get this working.

elektronep, 2020-04-28:
windows=(window)

window_positions=(
"0 0 1023 767" \
)

camera_names=(camera)

camera_feeds=( \
"rtsp://USER:PSW@10.0.0.10:554/mpeg/media.amp" \
)

elektronep, 2020-04-28:
This was my final config. Please make it clear in the readme that < > are not to be included in the URL. Thanks for your help.

closed 2022-02-16 by Anonymousdog