Anonymousdog/displaycameras #49: Touchscreen / keyboard support

nicolake, 2020-04-30:
Hi, I'm using this solution on my raspberry pi 3 + 7inch touchscreen and works very well.
I think it would be great to add support to the touchscreen and keyboard to navigate through the different cameras.
e.g.

- When is on grid mode, touching a camera stream should open on full screen. A second touch on the screen brings back the grid view.

- (nice to have) If rotation is active, when swipe to the left or right change the screen. 

I did some research and found  'evtest', that is a tool that log the input of any device, but I don't know if it is the best solution for this task.

Thanks!!

Anonymousdog, 2020-04-30:
RpiSurv may do some of those things all of which are well outside the
targeted use case for this project.  This project is for unattended viewing
(with no operator/viewer/user interaction with the displays); it is
intended as an improvement that adds proper service registrations and use,
a repair function, separation of data from code, and modularity to an older
shell script.  To employ interactive viewing, it would have to be rewritten
in an object-oriented manner and language.

Sorry,
Andy

SvenVD, 2020-05-03:
For reference: the touchscreen code for basic rotation control is now included in v2.1.8 of rpisurv. 
https://github.com/SvenVD/rpisurv/releases/tag/v2.1.8

closed 2021-08-18 by Anonymousdog