Anonymousdog/displaycameras #51: Can you add the camera patrol function.Thank you very much for your dedication.

wjq8811, 2020-05-16:


Anonymousdog, 2020-05-17:
No idea what you mean by that.

wjq8811, 2020-05-20:
With the 3x3 template, only 9 screens can be displayed. If there are 10 cameras, you can use one of them to display two cameras at the same time, and set it to switch once every 30 seconds. If there are 12 cameras, you can use 3 of them to display 2 cameras at the same time, and set 30 seconds to switch once.

I don't know if it can be expressed clearly.

Anonymousdog, 2020-05-20:
The rotation loops and math were the hardest part of this to get working.
I don't know if it's even possible to do what you're asking without an
object oriented language.  RPi Surv is likely better suited for your
purposes.

wjq8811, 2020-05-20:
well. Thank you very much. I wish you a happy life.

closed 2022-02-16 by Anonymousdog