Anonymousdog/displaycameras #57: videostream.cgi from Logilink WC0030W stays dark

sslneve, 2020-06-30:
The URL of the feed in the layout.conf.default is

"http://192.168.1.174/videostream.cgi?user=xxxx&pwd=xxxx"

This works in the browser but stays dark with displaycameras/omxplayer.
For another camera model, exactly this URL (different IP of course) does work with displaycameras.

Console output (feed is named "Chicken"):

Starting omxplayer for Chicken
Waiting for Chicken omxplayer startup 0
Waiting for Chicken omxplayer startup 1
Waiting for Chicken omxplayer startup 2
Waiting for Chicken omxplayer startup 3
Waiting for Chicken omxplayer startup 4
Waiting for Chicken omxplayer startup 5
Chicken failed playback
Running a repair on failed feeds.
Starting omxplayer for Chicken
Waiting for Chicken omxplayer startup 0
Waiting for Chicken omxplayer startup 1
Waiting for Chicken omxplayer startup 2
Waiting for Chicken omxplayer startup 3
Waiting for Chicken omxplayer startup 4
Waiting for Chicken omxplayer startup 5
Chicken failed playback

Anonymousdog, 2020-07-10:
When you try the debug instructions for that feed with omxplayer, what
happens?

sslneve, 2020-07-10:
pi@openWB:~ $ sudo omxplayer --no-keys --no-osd --avdict rtsp_transport:tcp http://user:pwd@192.168.1.176/videostream.cgi --live -n -1 --timeout 30
Video codec omx-mjpeg width 640 height 480 profile 192 fps 25.000000
Subtitle count: 0, state: off, index: 1, delay: 0

.. but there should be another line after this like:

V:PortSettingsChanged: 640x480@25.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0

which just does not appear.

Anonymousdog, 2020-07-10:
Try it without rearranging your URL from the first email.

sslneve, 2020-07-10:
Sorry, the IP has changed to .176

If I use the previous format, I get an unauthorized (only by adding the -i parameter):

pi@openWB:~ $ sudo omxplayer -i --no-keys --no-osd --avdict rtsp_transport:tcp http://192.168.1.176/videostream.cgi?user=xxx%26pwd=xxx --live -n -1 --timeout 30
[http @ 0x16aaa00] HTTP error 401 Unauthorized
have a nice day ;)

I had to URL-encode the "&" as otherwise it doesn't work at all.

Anonymousdog, 2020-07-10:
Not sure how to proceed from here.  Omxplayer had to be able to play the
feed for it to work within the displaycamerad package.

Andy

sslneve, 2020-07-10:
Yeah ... probably some incompatibility between cheap China crap camera and omxplayer...

closed 2020-07-10 by sslneve