Anonymousdog/displaycameras #60: 7 windows

wifispray, 2020-07-24:
Is it possible to create a 7 camera window like attached? See the examples you have but couldnt find one for this?

wifispray, 2020-07-24:
[image]

Brickabrak, 2020-07-24:
You can, just edit your layout.conf file to parse up the screens and make 7 variables to hold the screens. 

# 2x2 screen with 7 windows
windows=(upper_left upper_right_UL upper_right_UR upper_right_LL upper_right_LR lower_left lower_right)
# Make sure to account for each window above in the list below.

# Windows positions

window_positions=(
#First Row
#upper_left
"0 0 960 540" \
#upper_right
"960 0 1440 270" \
"1440 0 1920 270" \
"960 270 1920 540" \
"1440 270 1920 540" \

#Second Row
#lower_left
"0 540 960 1080" \
#lower_right
"960 540  1920 1080" \

)

# Camera Names

camera_names=(NE SE South SW West Vestibule)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds

camera_feeds=( \
"<Camera stream URL1>" \
"<Camera stream URL2>" \
"<Camera stream URL3>" \
"<Camera stream URL4>" \
"<Camera stream URL5>" \
"<Camera stream URL6>" \
"<Camera stream URL7>" \
)

If you've having issues, it's probably from the screens being too high of a resolution for the device. Try requesting a much lower resolution stream in the camera URL.

wifispray, 2020-07-29:
[image]
On the group of 4 cameras the 3rd camera (second row, 1st camera) seems to strech into camera 4 causing a flicker. Also not seeing anything in feed on bottom right corner. Config below:

# Large Upper Left with two to the right and three below (like the UBNT layout)
windows=(upper_left upper_right_UL upper_right_UR upper_right_LL upper_right_LR lower_left lower_right)
# Make sure to account for each window above in the list below.

# Windows positions
window_positions=(
#First Row
#upper_left
"0 0 960 540"
#upper_right
"960 0 1440 270"
"1440 0 1920 270"
"960 270 1920 540"
"1440 270 1920 540" \

#Second Row
#lower_left
"0 540 960 1080"
#lower_right
"960 540 1920 1080" \

)

# Camera Names

camera_names=(SE NE South West Vestibule SW)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds

camera_feeds=( \
"rtsp://172.16.201.11:7447/5f16b13c2548a6a0a947ea8e_1"
"rtsp://172.16.201.11:7447/5daf2378207dcbbc62ea7d61_1"
"rtsp://172.16.201.11:7447/5ec2899a254802adc9f7512c_1"
"rtsp://172.16.201.11:7447/5da5cae8207dfa7c43866fc2_1"
"rtsp://172.16.201.11:7447/5d1f61ee207d1fa8425db8ae_1"
"rtsp://172.16.201.11:7447/5d260899207da9fc0690e048_1"
"rtsp://172.16.201.11:7447/5dbd7ecf207d9787c719ba19_1"

Brickabrak, 2020-07-29:
You can add some buffer of 1 or 2 pixels between windows if you're worried about bleed, or you want to make a visible border around the streams. Feel free to change the coordinates. The format is just StartX, StartY, FinishX, FinishY. You can also play them manually with "omxplayer 'your_stream_here' --live --win 'StartX, StartY, FinishX, FinishY'.

For any camera that is blank, you should first test the stream is good in something like VLC or on the Rpi with "omxplayer 'your_stream_here' --live

If it works on it's own, then what's likely happening is the GPU can't support more information. You've probably already increased the gpumem in /boot/config.txt, so the next fix would be to request lower resolution streams from the cameras in the URL if that's possible, or setup the camera output streams to be lower resolution. 

Hope that helps!

wifispray, 2020-07-30:
Thanks. Really helpful. I managed to get the last stream working based on your help! :)

The coordinates seem to be the issue as the camera is streched over the bottom two windows of the small 4 cameras. It gives the appearnce or flashing picture becuase the camera 4 is overlayed on top of the streched camera 3.

Just need to figureout what the coordinates relate to :)

closed 2021-08-18 by Anonymousdog