Anonymousdog/displaycameras #61: Portrait Mode

akoryuerek, 2020-08-03:
Hi is there a solution to show the streams in portrait mode?

Brickabrak, 2020-08-03:
There are a couple of ways to do it. 
You can add the line "display_rotate=1" or "display_rotate=3" to the /boot/config.txt file. This will rotate the screen's orientation, so the coordinate system will change. 

Or, you can change line 117 of the "displaycameras" file, and add the the list of arguments "--orientation 90". Do this before the ">/dev/null &" part of the line. That will rotate the window that omxplayer is running. 

Or, you can request a rotated stream from the camera, and modify your URL in the layout file. You'll have to find the syntax for the URL, and the camera has to be capable of processing the request, but the benefit is it wont tax the GPU with extra processing like the first 2 ways will.

makoav, 2020-08-18:
I implemented the second suggestion of adding the --orientation 270 in my case to line 117.  This works when initially run but I have seen later I see a view randomly revert to no change in orientation.  I wonder if there was a failure in the stream and when it re-establishes the orientation is not set in this case.  Please advise on your thoughts?

b1tzmast3r, 2020-08-18:
I'd recommend using the boot config option.  Worked great for me!

makoav, 2020-08-18:
I would but this is not the only thing running on screen and changing boot/config will have global ramifications.  I need to address the OMXPlayer windows only

Anonymousdog, 2020-08-24:
There would be two places in the main script to modify options in
omxplayer. Look further down in the file.

Thanks,
Andy

closed 2022-02-16 by Anonymousdog