Anonymousdog/displaycameras #62: Screen image changed size after latest raspbian update

kw160008, 2020-08-04:
Last week when i updated to two of my raspberries they both now show video images only on part of the screen.  It has worked fine for the last year.

Here is how the cameras used to look:
[image]

Here is how the look now:
[image]

I did ensure that my systems are up to date with the following: 

To ensure all is update, run the following in a terminal, `sudo apt-get update && sudo apt-get dist-upgrade -y && sudo apt-get auto-remove -y && sudo apt-get autoclean`

**Display Configuration
 - Display/Monitor resolution: 1360x768
 - Window matrix:  2x2

**Camera information (please complete the following information):**
 - Number of cameras: 4
 - Camera models (if not Ubiquiti): G3 Flex Cameras

**Features enabled (please complete the following information):**
 - Rotation: [yes/no] no
 - Display Blanking: [yes/no] 
 - Display Detection: [yes/no]

**NVR information (please complete the following information):**
 - NVR hardware Cloud Key Gen2
 - NVR UniFi Video server version: Controller 1.13.4

**Additional context**
happens on both raspberries and happend at the same time.  The only thing changed, was updated the rasberry os.
**Files
Attach any suspect config files.

# Window names

# 2x2 screen with 3 off-screen windows
windows=(upper_left upper_right lower_left lower_right off_screen off_screen2 o$
# Make sure to account for each window above in the list below.

# Windows positions

window_positions=(
#First Row
#upper_left
# 640x360
"0 0 639 359" \
#upper_right
"640 0 1279 359" \

#Second Row
#lower_left#First Row
#upper_left
# 640x360
"0 0 639 359" \
#upper_right
"640 0 1279 359" \

#Second Row
#lower_left
"0 360 639 719" \
#lower_right
"640 360 1279 719" \

#off-screen
# 640x360 window just off-screen to the right
"1280 0 1919 359" \
# 640x360 window just below the other
"1280 360 1919 719" \
"1920 0 2559  359" \

#Camera Names

camera_names=(FrontDoor Showroom Showroom2 Register)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds
#Camera Names

camera_names=(FrontDoor Showroom Showroom2 Register)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds

camera_feeds=( \
# Mid-Res if your RPi can handle the load
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_1" \
# Low-Res otherwise
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_2" \
#FrontDoor
"rtsp://192.168.1.221:7447/lt6dxycmqH1dT6o8" \
#Showroom
"rtsp://192.168.1.221:7447/XtnOPgfmx5HOrNcK" \
#Showroom2
"rtsp://192.168.1.221:7447/VMl9RK7vmqBloPyH" \
#Register
"rtsp://192.168.1.221:7447/3qFwWqJDECIPLjlf" \

)

# Are we rotating cameras through the window matrix? (default false if not set $
# rotate="true"

Anonymousdog, 2020-08-04:
Since it was a system update that caused this, I would suspect you're no
longer running that display resolution.  It's probably 1440x900 now.

kw160008, 2020-08-04:
THANK YOU!  That fixed the issue!

closed 2020-08-04 by kw160008