Anonymousdog/displaycameras #63: Not a Issue but a Question

Therealjosephchrzempiec, 2020-08-05:
Hello i wonder if it is possible to Stream the camera to a webpage on the system that is displaying on the monitor. In my case I have a raspberry pi That I'm displaying right now a Single camera on the monitor. I'm also running lampp webpage server. I would like to stream my camera on a webpage served by the same system. Is this possible and How can i do this? I'm not a programmer by any means I'm learning as i go along.

Joseph

Anonymousdog, 2020-08-05:
I don't think this project is a good match for that. Your needs are well
outside the scope of what this does.

You're probably looking for Apache modules that would serve up the RTSP
feed without another package being involved.

Therealjosephchrzempiec, 2020-08-05:
Hello  Anonymousdog, l have looked up a few things and i couldn't find anything to get to work. Might be my lack of programming skills that is stopping that. Also most of these sites are not very good at detailing information on how to make it work. I'm using your code because it is the first one i got to work where the information i could understand.

Joseph

Anonymousdog, 2020-08-06:
This package and a javascript media player would work.

https://github.com/Roverr/rtsp-stream

Andy

Therealjosephchrzempiec, 2020-08-06:
Hello thank you i will check it out.

closed 2022-02-16 by Anonymousdog