Anonymousdog/displaycameras #64: What example to use for 4 cameras?

Therealjosephchrzempiec, 2020-08-05:
Hello I only have four cameras but i do not know what example to use to make 4 of my cameras work? I found the rotating one that does one at a time a little strange using it but it does work. Howver I'm trying to Display 4 even cameras as one time and i have no clue. Can someone please help me?

Joseph

Brickabrak, 2020-08-06:
You can use the same URL 4 times in the layout file to assign that URL to each camera you defined. 
or you can find some public IP cameras. 
Personally I use these for testing:
Low resolution factory potato camera
http://97.91.74.198/mjpg/video.mjpg

Public road intersection
http://166.140.164.188/mjpg/video.mjpg

1080p resolution resort camera
http://162.245.149.145:80/mjpg/video.mjpg

1080p boardwalk 
http://208.139.200.133/mjpg/video.mjpg

low resolution earth's rotation study
http://131.173.8.23:80/mjpg/video.mjpg?timestamp=1569451758585/

There are lots of public cameras if you search for them.

Therealjosephchrzempiec, 2020-08-06:
Hello displaying the cameras are not a problem. I'm trying to find a Quad view of your code in your example but Unsure which one i can use to make that type of view?

Joseph