Anonymousdog/displaycameras #65: Odd issue with rotated 1x3 display

NicholasBoccio, 2020-08-06:
**To Reproduce**
Steps to reproduce the behavior:
1. Install per README.md
2. Configure my feeds in layout.conf.default
3. Run 'sudo displaycameras start'
4. See error "..."

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Raspberry Pi (please complete the following information):**
 - Hardware model
 - OS: [e.g. Raspbian_Lite Stretch]
 - Console: [X-session/Pixel vs Text Console]
 - gpu memory split: [e.g. 256MB]
 - displaycameras Version [e.g. 0.8.3.3]

**Display Configuration
 - Display/Monitor resolution: [e.g., 1920x1080]
 - Window matrix: [e.g. 2x2]
 - Feed resolution: [e.g., low-resolution or 640x360]

**Camera information (please complete the following information):**
 - Number of cameras: [e.g. 6]
 - Camera models (if not Ubiquiti):
 - Managed/Unmanaged

**Features enabled (please complete the following information):**
 - Rotation: [yes/no]
 - Display Blanking: [yes/no]
 - Display Detection: [yes/no]

**NVR information (please complete the following information):**
 - NVR hardware
 - NVR UniFi Video server version: [e.g., 3.9.7]

**Additional context**
Add any other context about the problem here.

**Files
Attach any suspect config files.

Loss of synchronization & bad lag once 3rd stream starts in 1x3 configuration (rotated portrait mode).

I have been using this for 1 year, no issues, and recently all 3 of my displays immediately become laggy (within 2 seconds of the reboot and 3rd stream loading, the .5 second delay turns into 1-2 second delay and continues to get worse as it stays on. It does not recover.

Oddly, the same Pi, same code (just changed the rotation and layout) it will remain stable in a 2x2 grid.

I cannot figure out what happened. I thought Ubiquity might have tried to sabotage RTSP streaming, so I removed the cameras from Unifi and downgraded the firmware - no change. I then factory reset the cameras - no change. 

My only experience with raspberry pis is this project, so I don't know what else to trouble shoot. I updated 1 pi - no change, and then I completely started over and again no change. 

To reproduce:
set rotation in boot.txt to 1
screen blanking (doesnt matter if on or off)
No to display detect
Edit layout config as such:
"0 0 1080 640"
"0 640 1080 1280"
"0 640 1080 1920"
 [image]

All 3 cameras RTSP streams are low quality 
1st is from G4 Pro
640x360
2nd and 3rd are from G3AF
640x360 each.

I've been fighting this for 1 week, trying new Pis, new SD cards, and changing wifi access point. Nothing has changed the issue. Thanks for the help

akoryuerek, 2020-08-06:
Look here
https://github.com/Anonymousdog/displaycameras/issues/61#issuecomment-668052526

and don´t rotate in config.txt. Use --orientation 90 in displaycameras

NicholasBoccio, 2020-08-06:
> Look here
> [#61 (comment)](https://github.com/Anonymousdog/displaycameras/issues/61#issuecomment-668052526)
> 
> and don´t rotate in config.txt. Use --orientation 90 in displaycameras

Thank you for the reply, however this did not rotate the display. Can you include a screen shot of the code?

NicholasBoccio, 2020-08-06:
Also, my display is rotated fine, the streams just lag after the 3rd stream starts.

However, in a 2x2 display (same streams, same quality) there is no issue. 

Again, these have worked fine for a year, I dont know what has changed.

NicholasBoccio, 2020-08-06:
I think I figured it out. I think that with a recent Ubiquiti update, that their camera firmware upgraded the FPS or bitrate of the cameras. The displays behave normally if I reduce the quality to 10 FPS and 1000bps, which is fine for what I use these for.

I think that because I have the 1x3 portrait mode, that the Pi is working more to stretch the 3 streams, than the other pi since the 2x2 display requires practically no stretching or resizing.

I'd love to get feed back on this, and see if it makes sense to everyone that is more familiar with the player and the code than I am.

Anonymousdog, 2020-08-08:
This layout,
Edit layout config as such:
"0 0 1080 640"
"0 640 1080 1280"
"0 640 1080 1920"
Overlaps the 2nd and 3rd windows. This will reliably cause display problems
as soon as the last stream loads.

Thanks,
Andy

NicholasBoccio, 2020-08-08:
I apologize, that was a typo. Here is what the 1x3 display was configured as: 
[image]

As I mentioned, I believe the new forced FW caused the RTSP streams to default to 30fps each, do you think that, coupled with the stretching of the 640x360 stream could have degraded the performance in the pi, when the 2x2 displays continued to display properly?

Anonymousdog, 2020-08-09:
Sounds reasonable to me. You can't reduce bitrate since the update?

closed 2022-02-16 by Anonymousdog