Anonymousdog/displaycameras #66: More than 4 feeds on 1 monitor?

sabot54, 2020-08-06:
I am wondering if I can have more than 2x2 layout using this on a pi4. I am trying to setup a 4x4 on one monitor. Is this possible already?

NicholasBoccio, 2020-08-06:
You can try it out pretty easily by adjusting the layout file - did you want all 16 cameras to be the same size?

sabot54, 2020-08-06:
Thanks I am trying out now @NicholasBoccio 
Have you ever seen a touchscreen function? Sorry I am pretty new and I tried to read the README but I might have done more skimming than actual reading.

Anonymousdog, 2020-08-08:
15 displays appears to be a omxplayer limitation. If exhaustively optimized
(with smallest, lowest bit rate feeds and carefully chosen gpu_split), you
may get 15 5o display reliably.

Andy

closed 2022-02-16 by Anonymousdog