Anonymousdog/displaycameras #68: remote rtsp camera?

Therealjosephchrzempiec, 2020-08-10:
Hello i was wodering if i have a camera in my office and i wanted to watch it at home. Can i use my external ip address to view it on my raspberry pi? I have one locally in my office that can see the camera but i want to watch it at home. I did Port forward the rtsp port address.

Joseph

Anonymousdog, 2020-08-11:
That should work fine. Test the feed first with VLC or omxplayer to
validate the port forwarding.

Thanks,
Andy

Therealjosephchrzempiec, 2020-08-11:
Is there any other port I need besides the rtsp port?

Therealjosephchrzempiec, 2020-08-11:
Any other port needed to be added?

Anonymousdog, 2020-08-12:
Nope

Therealjosephchrzempiec, 2020-08-12:
Hello Thank you that seem to work. I had some kind of hickup where it wouldn't let me stream from my camera at my office to my home. but a reboot of my modem and it seems to work perfectly now. I do have static ip address on my network so that will never change.

Joseph

closed 2022-02-16 by Anonymousdog