Anonymousdog/displaycameras #71: Unable to change order of camera or add

sm2b, 2020-08-10:
Great program!

Question 
2x2 trying to change the order of a camera it will accept writing it in the "layout.conf.default" but doesn't actually change in the display.

Then tried for a 5 camera setup 2 top three bottom I couldn't get that to display as well.

So I started from scratch reformatting reinstalling, started with the 2x2 again this time I got one camera to display and a second camera trying to flash in.

all this is new to me so I'm sure i may be missing something simple.

The help is appreciated!

Anonymousdog, 2020-08-11:
Make sure you're editing the working file in /etc/displaycameras/ folder.
If that's not the problem, post a redacted version of your layout file.

sm2b, 2020-08-12:
Confirm working in /etc/displaycameras folder, did not touch the "layout.conf.default".  Also I am pushing this to a 4k tv and it works 2x2 but can't change order.  Also if I change the resolution as in the file it breaks. I wasnt able to get the 5 camera (2 on top 3 bottom) to work at all.

thanks!
[Original 2x2 redacted.txt](https://github.com/Anonymousdog/displaycameras/files/5066429/Original.2x2.redacted.txt)

Anonymousdog, 2020-08-13:
Unlese you change something the system almost always uses the default
layout file.

closed 2022-02-16 by Anonymousdog