Anonymousdog/displaycameras #72: Run cameras in background of apps?

Therealjosephchrzempiec, 2020-08-13:
Hello is it possible to run the cameras behind the apps that i run on the raspberry pi? So say i need to reboot and i push the start menu button i can not see the menu because the camera are on top of everything. I did move the cameras down so the menu bar is showing. But i can not see any part of the menu it goes behind the cameras.

joseph

Brickabrak, 2020-08-13:
The only thing I can think of is you can change the player command. (Line 117 of /usr/bin/displaycameras) and add "--alpha xxx" to the line before the >/dev/null part. 
Without that argument, all video is at 255 alpha, or full opacity. You can lower this value to make videos transparent, and if you have a desktop behind them it could be seen better.

Anonymousdog, 2020-08-13:
This package is intended for standalone RPis that do one thing: display
video streams. That's explicitly in the README.MD file. It is outside
project scope to support interactive use of the RPi.

Therealjosephchrzempiec, 2020-08-13:
The problem I'm facing is that even if i put the cameras in smaller screens they still ontop of any application i run on the raspberry pi. I just need the cameras to be behind everything.

Joseph

Anonymousdog, 2020-08-14:
Correct. That is the intent. It is not made for an RPi used interactively
or performing other tasks (especially video display tasks).

Therealjosephchrzempiec, 2020-08-14:
Understandable thank you. What I'm trying to do is design my own home automation setup at my house  displaying my office cameras and home cameras plus some temperature sensors. Sense I only have a 4 camera's i was trying to do some overlaying on them. Or maybe a button or two for rebooting the pi or shutdown. But this is a good start thank you.

Anonymousdog, 2020-08-14:
Answers would probably involve display layers (as alluded to earlier).

Andy

horsesNhalo, 2021-12-18:
I'm about a year late but I just wanted to add since I've been at this for hours and figured it might help someone down the line. I use my pi for other purposes but when I'm not using I want to have displaycameras going. When it's fullscreen I can't control without SSH and I ain't doin that. So I have it setup now where I can stop and start the service, but also I set my background to white, removed all Icons and lowered the taskbar size, and using what Brickabrat mentioned set --alpha 128 in that player line right before the --timeout. Important note if you do this, there's 2 of these lines, and you'll need to set alpha in both or it won't work when you reboot.

closed 2022-02-16 by Anonymousdog