Anonymousdog/displaycameras #73: Playing Streams Higher than Display Res

photomatix18, 2020-08-13:
I can't seem to get anything to play from my camera that's higher than 1920x1080(same as my display). Is there an override switch for this somewhere?

Anonymousdog, 2020-08-13:
You really don't want to do that (even if you think you do). Downscaling
concerns are in the README.MD

photomatix18, 2020-08-13:
I want to record to my DVR in 4k, but the substream is only 480p max which
is too low to monitor.

photomatix18, 2020-08-13:
I'm getting these errors when playing it in omxplayer standalone. 

`[rtsp @ 0x195c660] max delay reached. need to consume packet
[rtsp @ 0x195c660] RTP: missed 231 packets
[h264 @ 0x195f780] error while decoding MB 12 37, bytestream -6
[h264 @ 0x195f780] concealing 23557 DC, 23557 AC, 23557 MV errors in I frame
[rtsp @ 0x195c660] max delay reached. need to consume packet
[rtsp @ 0x195c660] RTP: missed 259 packets
Input #0, rtsp, from 'rtsp://10.0.100.10:554/H264/ch1/main/av_stream':
  Metadata:
    title           : Media Presentation
  Duration: N/A, start: 0.733000, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 3840x2160 [SAR 1:1 DAR 16:9], 15 fps, 15 tbr, 90k tbn, 30 tbc`

Does that sound like it's the decoder or something to do with the camera? It works fine in VLC.

Anonymousdog, 2020-08-14:
No idea. That's a problem between your stream source and omxplayer.

Sorry,
Andy

closed 2022-02-16 by Anonymousdog