Anonymousdog/displaycameras #74: What about using vlc player instead of omxplayer?

akoryuerek, 2020-08-13:


Anonymousdog, 2020-08-24:
The future is vlc since omxplayer won't be in future releases of Raspbian
(IIRC), but I haven't done any of the work necessary to port this over to
vlc command line.

Andy

jeremy-walker, 2020-09-21:
Just curious, do you have any plans on porting to VLC? 

I have a 21 camera setup and omxplayer refuses to work beyond 16 cameras. Up to 16 works absolutely fine. Starting the 17th feed, it keeps throwing `Failed to connect to TV service: -1`.
```
Starting omxplayer for cam17
tvservice-client: Failed to connect to TV service: -1
Waiting for cam17 omxplayer startup 0
Waiting for cam17 omxplayer startup 1
Waiting for cam17 omxplayer startup 2
Waiting for cam17 omxplayer startup 3
Waiting for cam17 omxplayer startup 4
Waiting for cam17 omxplayer startup 5
cam17 failed playback
```
I'm running Debian Buster (10.4) on a Raspberry Pi 3B with 512mb allocated towards GPU RAM. Even with the 16 feeds running, my main memory usage is at 25% which means it's not a memory issue. 

I'm wondering if this is a limitation with omxplayer.

Anonymousdog, 2020-09-22:
Yes, this is covered elsewhere. This appears to be an omxplayer issue.

jeremy-walker, 2020-09-23:
So, any chance you might consider porting displaycameras to vlc?

SvenVD, 2020-12-30:
This is exactly what [Rpisurv 3](https://github.com/SvenVD/rpisurv/tree/v3_latest) is doing, Replaced omxplayer with vlc player.

closed 2022-02-16 by Anonymousdog