Anonymousdog/displaycameras #75: 3 x 3 Layout Issue

the-real-rob, 2020-08-20:
I have opened the layout file for 1920x1080 screen size and edited it with the 9 cameras. I have also tested each RSTP stream in VLC. 

The issue Im having is that only 4 cameras on the 9x9 grid are showing up, namely, grid slot 2,3,4,5. The remaining blocks are empty.

I copied the code from the 9x9 layout file into the default layout file.

Is this correct and what are the best methods to follow with regards to using other layouts (i.e. 9x9 / 3x2 etc)

Anonymousdog, 2020-08-24:
Please post a redacted version of your current layout file.

Andy

the-real-rob, 2020-08-24:
Here is the content from the file - appreciate your assistance! I have simply removed the rasp links, otherwise the file is the same as what I am currently using.

```
# This is the camera feed/windows layout configuration file for the
# displaycameras service.  It ONLY configures the layout and feeds for
# the cameras; the rest of the configuration is in displaycameras.conf.
# See the comments in that file for notes on configuring the below.

# Window names

# 3x3 640x360 on-screen 
windows=(upper_left upper_middle upper_right center_left center_middle center_right lower_left lower_middle lower_right)
# Make sure to account for each window above in the list below.

# Windows positions

window_positions=(
#First Row
#upper_left
# 640x360
"0 0 639 359" \
#upper_middle
"640 0 1279 359" \
#upper_right
"1280 0 1920 359" \

#Second Row (missing all but the far right window because large_left is double size
#center_left
# 640x360
"0 360 639 719" \
#center_middle
"640 360 1279 719" \
#center_right
"1280 360 1920 719" \

#Third Row
#lower_left
# 640x360
"0 720 639 1079" \
#lower_middle
"640 720 1279 1079" \
#lower_right
"1280 720 1919 1079" \

)

# Camera Names

camera_names=(Driveway FrontGarden1 FrontGarden2 Alleyway Fountain Pool UpPavement DownPavement DrivewayExterior)
#camera_names=(NE SE South SW West Vestibule NE2 SE2 South2 SW2 West2 Vest2)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds

camera_feeds=( \
# Low-Res
#Driveway
"rtsp://xxxx” \
#FontGarden1
"rtsp://xxxx" \
#FrontGarden2
"rtsp://xxxx" \
#Alleyway
"rtsp://xxxx" \
#Fountain
"rtsp://xxxx" \
#Pool
"rtsp://xxxx" \
#UpPavement
"rtsp://xxxx" \
#DownPavement
"rtsp://xxxx" \
#DrivewayExterior
"rtsp://xxxx" \
)

# Are we rotating cameras through the window matrix? (default false if not set here)
rotate="false"

```

Anonymousdog, 2020-08-25:
Are the sources higher resolution than 640 x 360? What were the results of
performing the debugging steps in the README.MD file?

Thanks

the-real-rob, 2020-08-26:
The individual cameras are full HD 1920x1080. I will dive into the readme.md file and follow the steps. Apologies for the delay in getting back to you.

Anonymousdog, 2020-08-26:
You'll never squeeze 9 full HD feeds onto a single display.  They need to
be the same or smaller resolution as their display windows.

Thanks

Brickabrak, 2020-08-26:
In my testing, I can fit at best 2x1080p streams on a 1080p monitor along with 2 low res streams (640x360). 
You could maybe get 4x 1080p streams on a 4k monitor, but compressing the stream is very taxing to the GPU of the pi.

the-real-rob, 2020-08-26:
I have been successfully using the default layout which is a 2x2 layout and the cameras have been performing great, no issues at all. I have only recently tried to increase the grid to 3x3 and thats where the issue arises where only 4 out of the 9 cameras in the 3x3 grid appear. the other 5 camera grid blocks are empty.

Brickabrak, 2020-08-27:
I'm glad you're able to get so many HD videos playing. The blank videos are (from my experience) places where there was no more GPU RAM available, so they can't be displayed. 
The cameras don't know the window size that was given, so won't adjust based on that. From my testing cameras will lower the quality and bitrate to respond to a limited bandwidth, but never the resolution.

closed 2022-02-16 by Anonymousdog