Anonymousdog/displaycameras #77: Streams freeze, constantly stop and start

sesmith96, 2020-08-28:
**Describe the bug**
I have 3 Wyze Cams running in RTSP mode. I have them all set to 360p. 

With my monitor in 1080p@60hz, the screen loses connectivity or goes black as soon as the third camera is about to come live or would constantly flash. 

With my monitor set to 720p@50hz, I can get all 3 cameras but with 3 streams running, 2 out of the 3 cameras constantly freeze, they go away, and then come back, freeze a minute or two later, etc. If I remove one camera, and just stream 2, it seems to almost work fine. Other players can get the streams without issue. 

**Raspberry Pi (please complete the following information):**
 - Hardware model: 3b+
 - OS: Latest lite
 - gpu memory split: tired 256 and 512
 - displaycameras Version: latest

**Display Configuration
 - Display/Monitor resolution: 1280x720
 - Window matrix: 2x2
 - Feed resolution: 360p setting in wyze

**Camera information (please complete the following information):**
 - Number of cameras: 3
 - Camera models (if not Ubiquiti): Wyze Cam

**Features enabled (please complete the following information):**
 - Rotation: no
 - Display Blanking: no
 - Display Detection: no

jonhassall, 2020-10-02:
Have you tried forcing TCP?

excellapay, 2020-10-10:
are you running over WIFI or wired. I have 4 up display running on 3B+ with 512 Split.  This will sound like an odd change, use a new MicroSD.  I had a Pi that ran Screenly.io that kept locking up.  The MicroSD was the issue.

sesmith96, 2020-10-11:
As it turns out, my Cheapo Wyze cams don't actually change resolution when you change it in the app, despite them saying it should, so I had 3 1080p streams. I have them rotating full screen now and it's working great.

closed 2020-10-11 by sesmith96