Anonymousdog/displaycameras #78: RPI4 Dual Monitors -

excellapay, 2020-10-06:
Has anyone gotten dual monitors in RPI4 to work.  with the GUI, I can have streams on 2 monitors, but I like this setup and would like to have it put 4 streams on each of the 2 monitors.  The "offscreen" is not going to monitor 2.

jlbucklew, 2020-10-08:
I am attempting the same setup. I can't get any streams to start. Is it the same for you?

excellapay, 2020-10-10:
Correct. Same for me.  Cannot get it to display on second.

Anonymousdog, 2020-10-14:
It appears that you have to specify the (second) monitor (rather than a
pixel space where you expect it to be relative to the first monitor) with a
"--display x" parameter in the omxplayer command line. Supporting that
would require a pretty thorough rewrite if we are still supporting
rotation, and, as this project is nearing the end of its useful life, I
don't intend to embark on such an effort.

Thanks,
Andy

excellapay, 2020-10-15:
Thanks for the effort you have put into this.  I understand.  Have you seen a different project that works on the Pi and can do dual monitors?  I am thinking that the Pi is a little under powered.

Anonymousdog, 2020-10-16:
RPiSurv may handle it

Andy

closed 2022-02-16 by Anonymousdog