Anonymousdog/displaycameras #84: Background

joachimvanhove, 2020-11-25:
Is it possible to add a background image behind all the streams.
I'd like to show a few cameras in a special layout, with a custom background (brand logo...)

Anonymousdog, 2020-11-26:
Replace the /usr/bin/black.png file with desired image file. Ensure the
name is the same (case sensitive). Ensure blanking is enabled.

Thanks,
Andy

cdaugherty82, 2021-01-06:
I'm curious about this as well. I have replaced the /usr/bin/black.png file with an all black 1920x1080 black png image. I have 2 cameras displayed on the center of my monitor. I'd like to have the right and left sides blacked out so i dont have to see the raspberry text next to the images.

Here's what my screen looks like.
[https://i.ibb.co/NC0J0w8/IMG-8860.jpg](url)

RUHavingFun, 2021-08-17:
I have set blank="true" and checked that fbi is there.  But I am still getting my desktop as they switch.  Not sure what to check or do to resolve.

Anonymousdog, 2021-08-18:
> I'm curious about this as well. I have replaced the /usr/bin/black.png file with an all black 1920x1080 black png image. I have 2 cameras displayed on the center of my monitor. I'd like to have the [...]

You'll need to use an image that is the same aspect ratio as your ultra-wide display (or edit it to be the same ratio). The other option is to edit the line in /usr/bin/displaycameras calling fbi to use width rather than height for autozoom: "fbi --noverbose -T 2 --fitwidth /usr/bin/black.png >/dev/null 2>&1 &"

Anonymousdog, 2021-08-18:
> I have set blank="true" and checked that fbi is there. But I am still getting my desktop as they switch. Not sure what to check or do to resolve.

My guess is, honestly, an installation not run per instructions. The blanking and fbi part of this system is incredibly minor. If you run interactively (have a remote ssh running from another machine in order to recover from a blank screen) `fbi --noverbose -T 2 /usr/bin/black.png >/dev/null 2>&1 &`, you should get a black screen. If you don't, something is very odd about your display/frame buffer setup, fbi is broken somehow, or the /usr/bin/black.png file is missing. To recover from black screen, `sudo pkill fbi` from remote ssh session.

closed 2021-08-18 by Anonymousdog

RUHavingFun, 2021-08-22:
So I changed to be 2x2 and not do any rotating.  If I power off my cams the cameras feed of course drops as would expect.  But the interesting thing is the background is all black.  So it seems the black.png is displaying, just not when rotating feeds.