Anonymousdog/displaycameras #85: 10 windows

wifispray, 2020-12-02:
Hi All,

As per this thread, can seem to get 10 window version working. What is the logic to getting the x coordinates?

https://github.com/Anonymousdog/displaycameras/issues/60

I want to have large upper left, 4 small upper right, lower left large and 4 small lower right.

So far I have this:
window_positions=(
#First Row
#upper_left
"0 0 960 540"
#upper_right
"960 0 1440 270"
"1440 0 1920 270"
"960 270 1440 540"
"1440 270 1920 540" \

#Second Row
#lower_left
"0 540 960 1080"
#lower_right

Cant work out the lower_right piece. any help would be great

Anonymousdog, 2020-12-06:
X coordinates are just like upper right.
Y coordinates just like that but shifted down 540 pixels.

0 0 959 539 (for 1st window), btw since 0,0 is the 1st pixel (not 1,1).
Otherwise, your windows are too big by one pixel in each direction and
overlap that same amount.

wifispray, 2020-12-08:
Thanks dude. So for the #lower_right (4 windows), what would that look like?

wifispray, 2020-12-14:
Can you help?

wifispray, 2020-12-21:
> X coordinates are just like upper right. Y coordinates just like that but shifted down 540 pixels. 0 0 959 539 (for 1st window), btw since 0,0 is the 1st pixel (not 1,1). Otherwise, your windows are too big by one pixel in each direction and overlap that same amount.
> […](#)

Can you help?

closed 2021-08-18 by Anonymousdog