Anonymousdog/displaycameras #86: 2x2 setup only displays upper left cam

davefrompeg, 2020-12-02:
I have a 2x2 setup on a raspberry pi 3b+   
There are four cameras setup which I confirmed are working via VLC
The upper left cameras works flawlessly but when adding any of the others...  upper right, lower left or lower right
the display goes blank and only recovers after i stop the displaycameras.service

The pi seems to work with a 1600x900 res so I edited the layout.conf.default to what I think are accurate sizes

davefrompeg, 2020-12-02:
window_positions=(
#First Row
#upper_left
# 960x540
"0 0 799 449" \
#upper_right
# 960x540
"800 0 1599 449" \

#Second Row (missing all but the far right window because large_left is double size
#lower_left
# 960x540
"0 450 799 899" \
#lower_right
# 960x540
"800 450 1599 899" \

Anonymousdog, 2020-12-06:
Post that whole file as an attachment. It's hard to decipher here.

closed 2021-08-18 by Anonymousdog