Anonymousdog/displaycameras #87: HTTP stream with/without credentials not working

fotisK69, 2020-12-03:
hi,

i have a raspberry pi zero W with the according raspberry camera that is using motion and everything seems to work fine since i am able to view from my laptop via the http-stream url the cameras streaming video.

So, in my case the url that i am using to view is:
http://<ip>:8081/?login=<user>&password=<passwd>

I am trying not to start adding this stream to your very nice application but it seems that something is going wrong.
Now, having said that, i am not sure if this is due to using http-stream instead of rtsp or due to any other configuration issues that i miss in displaycameras! Like video size or what ever else needs to be configured.

From the motion.config:
# Image width (pixels). Valid range: Camera dependent, default: 320
width 640

# Image height (pixels). Valid range: Camera dependent, default: 240
height 480

# The mini-http server listens to this port for requests (default: 0 = disabled)
stream_port 8081

So, the question is does displaycameras work with http-stream and credentials:
camera_feeds=( \
# Low-Res
#Driveway
"http://<ip>:8081/?login=<user>&password=<passwd>” \
)

In case this is not possible do have any suggestion how I can configure motion to stream http and rtsp (to different port I guess).

Thanks in adv.

Anonymousdog, 2020-12-06:
The README.MD contains debugging information. Use the interactive command
line omxplayer instructions to test the feed with omxplayer. The config
file looks right; you may have to rewrite a bit of the main script to
support http.

fotisK69, 2020-12-13:
Hi,

thanks for your time.
I got it working with this kind of setup:
http://<user>:<passwd>@<ip>/video/mjpg.cgi

The last part /video/mjpg.cgi depends on the camera and how it stream the video.

Thanks again

closed 2020-12-13 by fotisK69