Anonymousdog/displaycameras #90: Vcodec id unknown: ad

tonydm, 2020-12-15:
**OS:** Raspian Lite
```
pi@raspberrypi:~/displaycameras-0.8.3.3 $ cat /etc/issue
Raspbian GNU/Linux 10 \n \l

pi@raspberrypi:~/displaycameras-0.8.3.3 $ uname -a
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
```

**Ensure your operating system is up-to-date**
I have run the following and preformed a reboot
`sudo apt-get update && sudo apt-get dist-upgrade -y && sudo apt-get auto-remove -y && sudo apt-get autoclean`

_**If you do the above two things and can still replicate the issue, then...**_

**To Reproduce**
Steps to reproduce the behavior:
1. Installed per README.md: `pi@raspberrypi:~/displaycameras-0.8.3.3 $ sudo ./install.sh`
2. Configured my feeds in layout.conf.default
3. Made no changes to /etc/displaycameras/displaycameras.conf
3. Ran 'sudo displaycameras start' per documentation, also tried `sudo systemctl restart displaycameras.service`
4. I see no error when running sudo systemctl start|stop|restart displaycameras.service, returns to prompt.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Raspberry Pi (please complete the following information):**
 - Hardware model: 4 4GB
 - OS: Raspbian_Lite Stretch
 - Console: text Console
 - gpu memory split: 256MB - Accepted default on install.sh
 - displaycameras Version 0.8.3.3

**Display Configuration
 - Display/Monitor resolution: 1366x768
 - Window matrix: 1x1
 - Feed resolution: 1280x720P
 - Encoding: H.265

**Camera information (please complete the following information):**
 - Number of cameras: 2
 - Camera models: Annke C800 4K
 - Managed
 - - Capable of: Main-stream: 3840x2160/15fps and Sub-stream: 640x480/30fps

**Features enabled (please complete the following information):**
 - Rotation: yes
 - Display Blanking: no - Until I understand if I need to disable
 - Display Detection: no  - Until I understand if I need to disable

**NVR information - Don't have

**Additional context**
1: The rstp URL works in VLC
2: I don't understand what the `Vcodec id unknown: ad` means when running the command line test
3: I have no idea how to configure layout config to my 1366x768 display

`pi@raspberrypi:~/displaycameras-0.8.3.3 $ sudo omxplayer --no-keys --no-osd --avdict rtsp_transport:tcp  "rtsp://user:pass@192.168.1.254:554/channel=1&stream=1.sdp" --live -n -1 --timeout 30`
```
Vcodec id unknown: ad
have a nice day ;)`
```

### I don't understand why entrycam1 reports failed when it is the exact same rtsp feed as gatecam1.  I don't have my second camera installed yet so duplicated the rtsp stream URL for the setup
```
pi@raspberrypi:~/displaycameras-0.8.3.3 $ sudo systemctl status displaycameras.service 
● displaycameras.service - Displays camera feeds for monitoring
   Loaded: loaded (/etc/systemd/system/displaycameras.service; enabled; vendor preset: enabled)
   Active: active (exited) since Tue 2020-12-15 08:34:15 MST; 5min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 18315 ExecStart=/usr/bin/displaycameras start (code=exited, status=0/SUCCESS)
 Main PID: 18315 (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/displaycameras.service

Dec 15 08:34:12 raspberrypi displaycameras[18315]: Waiting for entrycam1 omxplayer startup 2
Dec 15 08:34:13 raspberrypi displaycameras[18315]: Waiting for entrycam1 omxplayer startup 3
Dec 15 08:34:14 raspberrypi displaycameras[18315]: Waiting for entrycam1 omxplayer startup 4
Dec 15 08:34:15 raspberrypi displaycameras[18315]: Waiting for entrycam1 omxplayer startup 5
Dec 15 08:34:15 raspberrypi displaycameras[18315]: entrycam1 failed playback
Dec 15 08:34:15 raspberrypi displaycameras[18315]: Camera Display Started
Dec 15 08:34:15 raspberrypi displaycameras[18315]: For complete status info, run
Dec 15 08:34:15 raspberrypi displaycameras[18315]: /usr/bin/displaycameras status.
Dec 15 08:34:15 raspberrypi displaycameras[18315]: Starting camera rotation
Dec 15 08:34:15 raspberrypi systemd[1]: Started Displays camera feeds for monitoring.

```

```
pi@raspberrypi:~/displaycameras-0.8.3.3 $ sudo /usr/bin/displaycameras status
gatecam1 is NOT playing
entrycam1 is NOT playing
```

**Files

### For testing the second camera (entrycam1) is the exact same rtsp feed as gatecam1
```
# Window names
# 1x1 screen (for 1024x600 screens)
windows=(on_screen off_screen)
# Make sure to account for each window above in the list below.

# Windows positions
window_positions=(
#First Row
#on_screen
# 1024x600
"0 0 1023 599" \
#off_screen
"1024 0 2047 599" \
)

# Camera Names
camera_names=(gatecam1 entrycam1)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds
camera_feeds=(
# Mid-Res
# gatecam1
"rtsp://user:pass@192.168.1.254:554/channel=1&stream=1.sdp" \
# entrycam1
"rtsp://user:pass@192.168.1.254:554/channel=2&stream=1.sdp" \
)

# Are we rotating cameras through the window matrix?  Set this explicitly to
# "false" if not desired in this display layout.
rotate="true"
```

SvenVD, 2020-12-15:
Support questions should be asked on https://community.rpisurv.net/

https://www.tapatalk.com/groups/rpisurv/viewtopic.php?p=16#p16 shows what we need for troubleshooting.

The error log is most probably in /var/log/daemon.log can you see if it is logged there?

Thanks

Anonymousdog, 2020-12-17:
>    4. Ran 'sudo displaycameras start' per documentation, also tried sudo
>    5. I see no error when running sudo systemctl start|stop|restart

Until you can figure out why the debugging command,  sudo omxplayer
--no-keys --no-osd --avdict rtsp_transport:tcp "rtsp://
user:pass@192.168.1.254:554/channel=1&stream=1.sdp" --live -n -1 --timeout
30, is failing, we can't really get any farther.

Andy

tonydm, 2020-12-18:
Hello,

With all due respect, your response is less than helpful.  Correct me if
I'm wrong, but you defined and wrote the code and debug routine.  So, as I
stated in my post, I added the logger.yml to the source code directory,
which by the way you do not specify anywhere in your README as to where to
place the file, only the file config itself.  I simple oversifte I
understand.  But an oversite none the less.  I happen to be a developer
myself and was able to make sense of it and where to place the code.  Don't
misunderstand me, I completely appreciate your contribution.  I too have
open source code on Github.  All I was asking for was some help as to why
my issue.  I tried to thoroughly document my install, platform, etc.  You
could ask for some additional info or suggest some other approach, etc.  So
a little assistance is a reasonable request I thought.  I've assisted ones
with questions regarding my contributed open source code on Github.  My
response has never been "`Until you can figure out why the debugging
command,...is failing, we can't really get any farther.`  It's a simple
command line use of the omxplayer, the very command you use in your code.
So I think a reasonable part of my issue post.

Thank you very much, and best

TD

Anonymousdog, 2020-12-18:
You think I am supposed to know the precise meaning of some random error
your use case (outside the scope of the project) produced? An error code
produced by a third party prerequisite software? You're asking me to figure
out why omxplayer doesn't successfully ingest your feed? ...and you're
indignant and entitled about the whole thing?!?! Truly stunning.

The project contains no logger.yml file; I have no idea what you're talking
about. The README is clear. Run the installer, edit the config files in the
/etc/displaycameras/ directory. There is no "placing files" anywhere.

What I wrote below is still true. Omxplayer has an issue with your feed.
Omxplayer is NOT my project. I don't support it, and it's not my
responsibility to figure out why it won't play your second feed. It may
have something to do with it's being some sort of duplicate of the first
feed; I don't know because it's a problem unique to your setup, and it's
not caused by the project code. The proof of that is in omxplayer's failure
to play your feed interactively (at the command line) with out its being
wrapped in my code.

I'm sorry that's frustrating  to you, but it's still not my responsibility.
That problem is between your feed and omxplayer and whatever the omxplayer
documentation says about the error message it's returning.

Levanterman, 2021-04-21:
Hi @tonydm I too had the same problem with the 'codec id unknown: ad ' error. I knew my RTSP urls were correct (they worked in VLC) but didnt work in omniplayer. Just keep trying with omniplayer directly , playing around with your RTSP URLs.

For instance my working VLC RTSP URL was : rtsp://admin:password@192.168.1.204:554/unicast/c5/s1/live but that gave me the codec error in omniplayer. I had to remove the last /LIVE and also change the subfeed (s1) to h264 (from h265) in the camera config, then this URL worked with omniplayer and instantly worked in DISPLAYCAMERAS afterwards.  Working URL : rtsp://admin:password@192.168.1.204:554/unicast/c5/s1/

Also my IP cameras had a password and my NVR had a different password. Ensure you are using the NVR username and password in the RTSP URL.

Thanks @Anonymousdog for posting works great :-)

Abrian, 2021-06-17:
I encountered the same problem today with the error message 'coded id unknown: ad'. The stream works perfectly in VLC but doesn't work in omxplayer. I agree with @Anonymousdog that the problem is out of his domain. Upon further investigation I found that the non-working stream's codec information was different from a stream that does work with omxplayer and displaycameras.

Searching for the error results in more information explaining that the encoding is not supported by omxplayer.

This let me to further investigate why this one camera stream is not working. Upon logging in to the problem camera's configuration, I noted that the stream was set as H.265 where the working camera was H.264. This was indeed to problem. I changed the video encoding setting to H.264 and everything started working correctly.

@Anonymousdog Well done on a great project and thank you for the time and effort you put into this. I am very grateful and using this on a daily basis.

Anonymousdog, 2021-08-18:
> I encountered the same problem today with the error message 'codec id unknown: ad'. The stream works perfectly in VLC but doesn't work in omxplayer. I agree with @Anonymousdog that the problem is out of his domain. Upon further investigation I found that the non-working stream's codec information was different from a stream that does work with omxplayer and displaycameras.
> 
> Searching for the error results in more information explaining that the encoding is not supported by omxplayer.
> 
> This let me to further investigate why this one camera stream is not working. Upon logging in to the problem camera's configuration, I noted that the stream was set as H.265 where the working camera was H.264. This was indeed the problem. I changed the video encoding setting to H.264 and everything started working correctly.
> 
> @Anonymousdog Well done on a great project and thank you for the time and effort you put into this. I am very grateful and use your tool on a daily basis.

I appreciate this feedback, Abrian. I'm befuddled that someone calling themselves a developer doesn't understand the dividing lines between and among open source projects (and that support here is literally being provided for free).

closed 2021-08-18 by Anonymousdog