Anonymousdog/displaycameras #93: Rotation of two cameras in just one slot

CyberDefCon8, 2021-02-19:
Hi there, great great great little project - love the fact I'm now able to visualise a number of Unifi cams on some low cost/old hardware! Thanks...

I've played around with the configs and can't quite achieve what I'm trying to do.

For one deployment, I have 5 cameras - keen to keep 3 of them in the same space constantly, and the have the 4th slot rotate between cam 4 and cam 5.

Is there an easy way to achieve that?

Anonymousdog, 2021-02-19:
Not with the current package.

Sorry,
Andy

CyberDefCon8, 2021-02-19:
Ok no dramas - thanks anyway and chapeau for the work so far!

closed 2021-02-19 by CyberDefCon8

SvenVD, 2021-02-20:
Something like this is possible with rpisurv with a little trick: https://gist.github.com/SvenVD/7b973cde2ce5dce6c8284606434e3a14 see example layout 6. The same trick can possible be used with displaycameras too?